abandonware/noble #2: Merge back to active upstream

rzr, 2018-12-19:
Relate-to: https://github.com/noble/noble/issues/656

This active is blocked by those pending issues/reviews:

https://github.com/noble/noble/pull/851
https://github.com/abandonware/noble/pull/1
https://github.com/noble/noble/issues/656

petersaints, 2018-12-19:
Hi, could you please also fork "bleno" (https://github.com/noble/bleno) which is part of the "noble" family and depends from "node-bluetooth-hci-socket".

rzr, 2018-12-19:
Just did this:
https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-0

Testing welcome

petersaints, 2018-12-19:
You forked noble, and thanks for that. I'm talking about "bleno". "noble" implements the BLE Central Mode, whereas "bleno" implements BLE Peripherals. They complement each other.

rzr, 2018-12-19:
sorry ;) 
It's there 
https://github.com/abandonware/bleno
Feel free to make PR, but make sure that changes are forwarded upstream first and well tracked 
(mimic my changes) or read those hints:
https://www.slideshare.net/rzrfreefr/tizen-upstreamcooptdc2014pcoval/5

rzr, 2019-04-15:
A new version has been released with related patches..

Relate-to: https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-2

@petersaints , Bleno co-maintainers are still welcome check:

* https://abandonware.github.io/

rzr, 2019-05-27:
Relate-to: https://github.com/noble/noble/issues/886