abandonware/noble #4: MacOS Mojave support

hannseman, 2018-12-27:
Hi! 

Thank you very much for taking the initiative to create a maintainable fork. I understand that this fork mainly was created for getting support for Node >=10 but another big issue that noble has is missing support for MacOS Mojave. This PR https://github.com/noble/noble/pull/840 fixes this and because of noble being unmaintained the PR author created a new package `noble-mac`: https://github.com/Timeular/noble-mac.

It would be preferable to only have one fork. Maybe https://github.com/noble/noble/pull/840 could be merged into this repository and start a discussion with the `noble-mac` author about making contributions to this repository instead?

hannseman, 2019-02-07:
Please see https://github.com/Timeular/noble-mac/issues/23#event-2115812974

rzr, 2019-02-07:
Hi @hannseman 

Best option would be that upstream is rebasing on abandonware fork. I have not osx device I could use, eventually one could setup a build bot to check any PR on this repo ?

hannseman, 2019-02-07:
You already have Travis CI configured to run the test suite on OSX.

rzr, 2019-02-08:
Oh yes osx is listed in:
https://github.com/abandonware/noble/blob/master/.travis.yml#L3

Please confirm, that what we have is working fine,
maybe we could do continuous delivery too,
I am used to jfrog's bintray , maybe other could be suggested ?

BTW the link in readme is wrong it should be:
https://travis-ci.org/abandonware/noble/jobs/
feel free to patch this
@hannseman 

May I suggest @Timeular to rebase your branch on @abandonware noble's master ?

PR #8 (2019-04-15): Native mac bindings
Rebase https://github.com/noble/noble/pull/840 on this repo and add latest changes from [Timeular/noble-mac](https://github.com/Timeular/noble-mac)

Fixes #4
closed 2019-04-15 by rzr