abandonware/noble #12: Hard node crash on MacOS Mojave

hannseman, 2019-04-18:
Crashes with the following traceback:
```
FATAL ERROR:  Cannot read property 'find' of undefined
 1: 0x1000660f9 node::Abort() [/usr/local/bin/node]
 2: 0x100066785 node::OnFatalError(char const*, char const*) [/usr/local/bin/node]
 3: 0x10006666c node::OnFatalError(char const*, char const*) [/usr/local/bin/node]
 4: 0x100049c79 napi_fatal_error [/usr/local/bin/node]
 5: 0x104a9b641 Napi::Error::Fatal(char const*, char const*) [/tmp/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 6: 0x104aa6c45 ThreadSafeCallback::Impl::static_async_callback(uv_async_s*) [/tmp/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 7: 0x100768676 uv__async_io [/usr/local/bin/node]
 8: 0x100777d41 uv__io_poll [/usr/local/bin/node]
 9: 0x100768b0b uv_run [/usr/local/bin/node]
10: 0x1000486a7 node::StartNodeWithIsolate(v8::Isolate*, node::IsolateData*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) [/usr/local/bin/node]
11: 0x100048271 node::StartNodeWithLoopAndArgs(uv_loop_s*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) [/usr/local/bin/node]
12: 0x100047f0b node::Start(int, char**) [/usr/local/bin/node]
13: 0x7fff6f6973d5 start [/usr/lib/system/libdyld.dylib]
14: 0x1
Abort trap: 6
```

```
$ node -v
v11.13.0
```

Running @abandonware/noble at version `1.9.2-2`.

Can be reproduced with the following:
```js
const noble = require("@abandonware/noble");
class Scanner {
  constructor() {
    noble.on("discover", this.onDiscover.bind(this));
    noble.on("scanStart", this.onScanStart.bind(this));
    noble.on("stateChange", () => {
      noble.startScanning([], true);
    });
  }

  onScanStart() {
    this.log.debug("Started scanning.");
  }

  onDiscover(peripheral) {
    console.log(peripheral);
  }
}
const scanner = new Scanner();
```

It does not crash when the `scanStart` event callback is bound to an anonymous function like this:
```js
noble.on("scanStart", () => this.log.debug("Started scanning."));
```

Maybe @geovie has some insight in this?

mrstegeman, 2019-04-18:
Why is your `scanStart` handler using `this.log` rather than `console`? `this.log` doesn't appear to exist.

hannseman, 2019-04-18:
@mrstegeman you're right. That was a bad case of copy and paste from the actual implementation where I encountered this. I've fixed the example now, thanks! (Note that this does not affect anything as that line is never run because of the crash).

mrstegeman, 2019-04-18:
So no crash anymore?

hannseman, 2019-04-18:
No crash by using:

```
js
const noble = require("@abandonware/noble");
class Scanner {
  constructor() {
    noble.on("discover", this.onDiscover.bind(this));
    noble.on("scanStart", this.onScanStart.bind(this));
    noble.on("stateChange", () => {
      noble.startScanning([], true);
    });
  }

  onScanStart() {
    console.debug("Started scanning.");
  }

  onDiscover(peripheral) {
    console.log(peripheral);
  }
}
const scanner = new Scanner();
```

mrstegeman, 2019-04-18:
Great! Can you go ahead and close?

rzr, 2019-04-18:
Si please close the issue if no more relevant

Eventually you can try to submit your example to upstream ?
That could be useful to have some CLI BLE utilities as part of package.
 
Relate-to: https://github.com/abandonware/noble/issues/9

hannseman, 2019-04-18:
@mrstegeman thanks for the feedback.

closed 2019-04-18 by hannseman

geovie, 2019-04-18:
@hannseman Yes any exception thrown in the event handler will abort the node process.

hannseman, 2019-04-18:
@geovie good to know, found the cause now in my actual implementation. Thanks for the help.