abandonware/noble #13: Travis builds have started to break for me

AlCalzone, 2019-04-23:
See logs:

https://travis-ci.org/AlCalzone/ioBroker.ble/builds/522936778?utm_source=github_status&utm_medium=notification

OSX is failing during compilation on Node 6, 8 and 10.
Windows is failing on Node 6, but that seems to be related to the msbuild version.

Any idea whats going on or if this is even the fault of this library?

rzr, 2019-04-23:
make: g++-4.9: No such file or directory

AlCalzone, 2019-04-24:
Hmm. I assume travis must have changed their images then. It used to work in the previous build.

AlCalzone, 2019-04-25:
Sorry for bothering you - this seems to be entirely travis-related.

closed 2019-04-25 by AlCalzone

rzr, 2019-04-25:
Eventually cross link to travis related issue if it's reported.

AlCalzone, 2019-04-25:
I "fixed" it by not specifying the compiler version on OSX and dropping Node 6 on Windows from the test matrix. Did not investigate further.