abandonware/noble #20: Trying to list available ports with Espruino fails.

manuelnaturalcycles, 2019-08-06:
I have tried to get some help in the espruino forums but we got stuck.

Im running OSX Mojave 10.14, Node 12.7.0 (also tried 10.10.0 and 8.6.0).

I modify the espruino source code to require @abandonware/noble instead of noble, i installed @abandonware/noble v1.9.2-3

When I run `espruino --list` it dumps:

```
FATAL ERROR:  Cannot read property 'indexOf' of undefined
 1: 0x100077a0b node::Abort() [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
 2: 0x100077b8f node::OnFatalError(char const*, char const*) [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
 3: 0x100077a21 node::OnFatalError(char const*, char const*) [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
 4: 0x10005839a napi_open_callback_scope [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
 5: 0x105c8eb1d Napi::Error::Fatal(char const*, char const*) [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/lib/node_modules/espruino/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 6: 0x105c9d801 ThreadSafeCallback::Impl::static_async_callback(uv_async_s*) [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/lib/node_modules/espruino/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 7: 0x10086a81c uv__async_io [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
 8: 0x10087ad7d uv__io_poll [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
 9: 0x10086aca6 uv_run [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
10: 0x1000ad491 node::NodeMainInstance::Run() [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
11: 0x100057abf node::Start(int, char**) [/Users/manuelabergcobo/.nvm/versions/node/v12.7.0/bin/node]
12: 0x7fff6eecb085 start [/usr/lib/system/libdyld.dylib]
Abort trap: 6
```

seems like something goes awry in step 5 or 6 of the stack trace, any clues?

its important to note that if i run some script to discover peripherals using @abandonware/noble I can actually see my pucks in the listing. so I can discover peripherals but I cannot connect to them because I cant even get the ports to list with `espruino --list`

I know that it looks like an espruino problem, but I have been trying 2010912093093 things with noble, noble-mac and @abandonware-noble but this package seems the most popular and maintained one, would love some help :(

rzr, 2019-08-06:
Hi thanks for report, do know an configuration where the issue is not happening (node-8?) ?

Feel free to cross link to espruino forum here 

cc: @gfwilliams

manuelnaturalcycles, 2019-08-06:
https://github.com/espruino/EspruinoTools/issues/63

manuelnaturalcycles, 2019-08-06:
i cant get this to work on any version of node, any particular version you recommend?

gfwilliams, 2019-08-06:
Hi - on EspruinoTools' GitHub you tried this with noble-mac and that's what we verified was working fine via the example - why did you now move to abandonware noble?

manuelnaturalcycles, 2019-08-06:
hi @gfwilliams , sorry that you got bothered here too, it was not my intention, i followed this link https://github.com/Timeular/noble-mac/issues/23 in which they reference this link https://github.com/abandonware/noble/issues/4 and thats how I guessed that abandonware/noble was more up to date than noble-mac.

the examples for scanning for peripherals show my pucks both with noble-mac and @abandonware/noble, but if i modify espruino to use noble-mac or @abandonware/noble, the --list command does not show any ports :S

im starting to give up on this and just use the web ide, as painful as that is, its still okay.

i have nothing but thanks to say to you gordon and all the people working to make this stuff happen :)

manuelnaturalcycles, 2019-08-14:
do you have any clues @rzr ?

rzr, 2019-08-14:
Not really, I am a bit busy on other tasks now, but I might try to setup @espruino on targets, which one do you recommand ? I have esp32 and a couple of stm32 ones.

manuelnaturalcycles, 2019-08-14:
not sure if i understand what you are asking!

gfwilliams, 2019-08-27:
I believe the issue is just related to Bluetooth LE on Mac, so the Thingy:52 from Fullstack should be good enough to see if you can see devices and connect on your Mac?

gfwilliams, 2019-11-18:
I believe this is actually now fixed. It wasn't really an issue with Noble per se.

Basically `dev.advertisement.serviceUuids` always seems to exist in Linux/Windows, even if it's empty. On Mac (and maybe this changed at some point?) sometimes it doesn't exist.

It got fixed by this commit in EspruinoTools:

https://github.com/espruino/EspruinoTools/commit/012c92db6b4090c4539471dd1f3e25076b6d9937

manuelnaturalcycles, 2019-11-20:
thanks @gfwilliams ! i will try it out again