abandonware/noble #21: node 12.8 - dependency @abandonware/bluetooth-hci-socket build fail

omayhemo, 2019-08-13:
I believe the reason this fork was made was to fix this problem in node 10.  Yet, I still seem to have this trouble with both the original noble and this @abondonware version.

```
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:196:23)
gyp ERR! stack     at ChildProcess.emit (events.js:203:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.19.57-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node" "--module_name=binding" "--module_path=/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/lib/binding" "--napi_version=4" "--node_abi_napi=napi" "--napi_build_version=0" "--node_napi_label=node-v72"
gyp ERR! cwd /home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket
gyp ERR! node -v v12.8.0
gyp ERR! node-gyp -v v5.0.3
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node --module_name=binding --module_path=/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/lib/binding --napi_version=4 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v72' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:203:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
node-pre-gyp ERR! System Linux 4.19.57-v7+
node-pre-gyp ERR! command "/usr/local/bin/node" "/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket
node-pre-gyp ERR! node -v v12.8.0
node-pre-gyp ERR! node-pre-gyp -v v0.12.0
node-pre-gyp ERR! not ok 
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node --module_name=binding --module_path=/home/pi/Public/scanner/node_modules/@abandonware/bluetooth-hci-socket/lib/binding --napi_version=4 --node_abi_napi=napi --napi_build_version=0 --node_napi_label=node-v72' (1)
npm WARN scanner@1.0.0 No description
npm WARN scanner@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @abandonware/bluetooth-hci-socket@0.5.3-1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @abandonware/bluetooth-hci-socket@0.5.3-1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```

marksyzm, 2019-08-13:
Yes but also primarily because it was abandonware

omayhemo, 2019-08-13:
Ya, I get that.  But the issue persists.  Is this forked maintained?  Can we expect it to get resolved?

rzr, 2019-08-13:
It is community maintained see:
https://abandonware.github.io/

mrstegeman, 2019-08-13:
This error actually exists in `@abandonware/bluetooth-hci-socket`, so we should transfer the issue there. The full error log is:

```
make: Entering directory '/workspace/node_modules/@abandonware/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
../src/BluetoothHciSocket.cpp: In static member function 'static void BluetoothHciSocket::Init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)':
../src/BluetoothHciSocket.cpp:129:82: error: no matching function for call to 'v8::FunctionTemplate::GetFunction()'
   target->Set(Nan::New("BluetoothHciSocket").ToLocalChecked(), tmpl->GetFunction());
                                                                                  ^
In file included from /root/.cache/node-gyp/12.8.0/include/node/node.h:63:0,
                 from /root/.cache/node-gyp/12.8.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/root/.cache/node-gyp/12.8.0/include/node/v8.h:5955:46: note: candidate: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/root/.cache/node-gyp/12.8.0/include/node/v8.h:5955:46: note:   candidate expects 1 argument, 0 provided
../src/BluetoothHciSocket.cpp: In member function 'void BluetoothHciSocket::poll()':
../src/BluetoothHciSocket.cpp:252:95: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                               ^
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function 'void BluetoothHciSocket::emitErrnoError()':
../src/BluetoothHciSocket.cpp:270:109: warning: 'v8::Local<v8::Value> v8::Object::Get(v8::Local<v8::Value>)' is deprecated: Use maybe version [-Wdeprecated-declarations]
   Local<Function> errorConstructor = Local<Function>::Cast(globalObj->Get(Nan::New("Error").ToLocalChecked()));
                                                                                                             ^
In file included from /root/.cache/node-gyp/12.8.0/include/node/v8-internal.h:14:0,
                 from /root/.cache/node-gyp/12.8.0/include/node/v8.h:25,
                 from /root/.cache/node-gyp/12.8.0/include/node/node.h:63,
                 from /root/.cache/node-gyp/12.8.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/root/.cache/node-gyp/12.8.0/include/node/v8.h:3419:55: note: declared here
   V8_DEPRECATE_SOON("Use maybe version", Local<Value> Get(Local<Value> key));
                                                       ^
/root/.cache/node-gyp/12.8.0/include/node/v8config.h:326:3: note: in definition of macro 'V8_DEPRECATE_SOON'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/BluetoothHciSocket.cpp:283:93: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
   Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                             ^
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE BluetoothHciSocket::BindRaw(Nan::NAN_METHOD_ARGS_TYPE)':
../src/BluetoothHciSocket.cpp:396:34: error: no matching function for call to 'v8::Value::IntegerValue()'
       devId = arg0->IntegerValue();
                                  ^
In file included from /root/.cache/node-gyp/12.8.0/include/node/node.h:63:0,
                 from /root/.cache/node-gyp/12.8.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/root/.cache/node-gyp/12.8.0/include/node/v8.h:2572:40: note: candidate: v8::Maybe<long int> v8::Value::IntegerValue(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int64_t> IntegerValue(
                                        ^~~~~~~~~~~~
/root/.cache/node-gyp/12.8.0/include/node/v8.h:2572:40: note:   candidate expects 1 argument, 0 provided
../src/BluetoothHciSocket.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE BluetoothHciSocket::BindUser(Nan::NAN_METHOD_ARGS_TYPE)':
../src/BluetoothHciSocket.cpp:418:34: error: no matching function for call to 'v8::Value::IntegerValue()'
       devId = arg0->IntegerValue();
                                  ^
In file included from /root/.cache/node-gyp/12.8.0/include/node/node.h:63:0,
                 from /root/.cache/node-gyp/12.8.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/root/.cache/node-gyp/12.8.0/include/node/v8.h:2572:40: note: candidate: v8::Maybe<long int> v8::Value::IntegerValue(v8::Local<v8::Context>) const
[... cut ...]
npm WARN workspace@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @abandonware/bluetooth-hci-socket@0.5.3-1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @abandonware/bluetooth-hci-socket@0.5.3-1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-08-13T17_43_38_105Z-debug.log
```

janhaa, 2019-09-16:
Original issue is here: https://github.com/noble/node-bluetooth-hci-socket/issues/84

PR #23 (2019-09-19): Update dependencies for Node 12.
Fixes #21
commit 070c9c61 (2019-09-19): Update dependencies for Node 12.

Fixes #21
closed 2019-09-19 by rzr