abandonware/noble #25: Impossible to get services when connect to a device

Ravaniss, 2019-09-19:
Hello everyone,

I'm depress right now, i hope someone can help me a little bit, i will really appreciate this... lol

My goal is simple :
  1 - On my first laptop, i'm using bleno
  2 - On my second laptop, i'm using nativescript-bluetooth v2.0.0-beta.23
  3 - When my second laptop read the service from my second laptop -> get a string

How can i do that ? So, time to explain a bit what i did so far ! :)

First laptop: I created a project with bleno

On my second laptop (i'm using VMWare with Ubuntu) i try to get the service from my first laptop. Big problem : The services are empty...

Here the full source code : 
https://gist.github.com/Ravaniss/3f4cf62e27acb9594204321052cc0b39

`
{ UUID: '74:29:AF:80:17:BA',
JS:   name: 'echo',
JS:   state: 'connected',
JS:   services: [ [length]: 0 ],
JS:   advertismentData: undefined }
JS: 'Peripheral connected with UUID: 74:29:AF:80:17:BA'
JS: 'Disconnected successfully'
`

You are going to say, why don't you use noble ? Because noble or bleno doesn't work on my second laptop using Ubuntu on VMware... Tried this a lot... But works perfectly fine on my first laptop using a SSD... (Should i buy a SSD only with Ubuntu and do a dual boot... ?)

I don't think than the problem comes to nativescript-bluetooth plugin, the problem is from bleno, how can i set a service and when i'm scanning, i get a services with a length of 1 ? x)

Ravaniss, 2019-09-22:
It's impossible to set a service... Can you help please ?

Here the code :
https://gist.github.com/Ravaniss/e0b398293a3f9c93784bbde0f99a1e5e

I'm using https://play.google.com/store/apps/details?id=com.manishtaraiya.bleterm

to scan and select a write / read / notify characteristic.

Ravaniss, 2019-09-28:
Hello,

I'm sorry but i'm buming this thread. ^^

Can you help with this please ? <3

Love

rzr, 2019-10-03:
May I suggest 1st to test this
noble/examples/advertisement-discovery.js

And try to reproduce using simplest configuration, preferably only with noble, bleno etc code

Ravaniss, 2019-10-03:
Will test it, thank you ! :) <3

Ravaniss, 2019-10-07:
So,

With one of my computer using bleno, I take the echo folder :
https://github.com/abandonware/bleno/tree/master/examples/echo

With my second computer using noble, i take :
https://github.com/abandonware/noble/blob/master/examples/advertisement-discovery.js

Scan OK, everything is fine, but i can't get any of my services, can't read, notify or write.

I think the problem is from bleno no?

Ravaniss, 2019-12-10:
Hello,

Any idea how to solve this problem please ? I'm still locked with this. ^^"

closed 2020-11-19 by Kursorr