abandonware/noble #30: Doesn't build with node 12.8.1 (electron 7.0.0)

jinpark, 2019-11-05:
If I try to use electron-rebuild with electron 7, it causes the build to fail.

I'm not sure if this is due to the changes in v8 for electron but it does seem to build on the "normal" (non electron) version of 12.8.1.

It doesn't seem to be the same problem as https://github.com/abandonware/noble/issues/21 but a problem with maybe node-usb.

```
$ node_modules/.bin/electron-rebuild
- Searching dependency tree
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@5.0.5
gyp info using node@12.8.1 | linux | x64
gyp info find Python using Python version 2.7.15 found at "/usr/bin/python"
gyp info spawn /usr/bin/python
gyp info spawn args [
gyp info spawn args   '/app/electron/node_modules/electron-rebuild/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/app/electron/node_modules/usb/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/app/electron/node_modules/electron-rebuild/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.electron-gyp/7.0.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.electron-gyp/7.0.0',
gyp info spawn args   '-Dnode_gyp_dir=/app/electron/node_modules/electron-rebuild/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/root/.electron-gyp/7.0.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/app/electron/node_modules/usb',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/app/electron/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from /root/.electron-gyp/7.0.0/include/node/v8-internal.h:14:0,
                 from /root/.electron-gyp/7.0.0/include/node/v8.h:27,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/root/.electron-gyp/7.0.0/include/node/v8config.h:343:49: warning: 'MicrotasksCompletedCallback' is deprecated [-Wdeprecated-declarations]
   declarator __attribute__((deprecated(message)))
                                                 ^
/root/.electron-gyp/7.0.0/include/node/v8.h:8513:3: note: in expansion of macro 'V8_DEPRECATE_SOON'
   V8_DEPRECATE_SOON("Use *WithData version.",
   ^~~~~~~~~~~~~~~~~
/root/.electron-gyp/7.0.0/include/node/v8config.h:343:49: warning: 'MicrotasksCompletedCallback' is deprecated [-Wdeprecated-declarations]
   declarator __attribute__((deprecated(message)))
                                                 ^
[... 188 lines cut ...]
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/app/electron/node_modules/electron-rebuild/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:203:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.15.0-1043-aws
gyp ERR! command "/root/.nvm/versions/node/v12.8.1/bin/node" "/app/electron/node_modules/electron-rebuild/node_modules/.bin/node-gyp" "rebuild" "--target=7.0.0" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd /app/electron/node_modules/usb
gyp ERR! node -v v12.8.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (/app/electron/node_modules/spawn-rx/lib/src/index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (/app/electron/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.error (/app/electron/node_modules/rxjs/internal/Subscriber.js:156:26)
    at Subscriber._error (/app/electron/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/app/electron/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/app/electron/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/app/electron/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/app/electron/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/app/electron/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.next (/app/electron/node_modules/rxjs/internal/Subscriber.js:143:22)
error Command failed with exit code 255.
```

jinpark, 2019-11-05:
As suspected, its a problem with node-usb . Building it within electron fails while building it outside electron succeeds (with node 12.8.1)

dsteinman, 2020-04-23:
In case you haven't got this working yet, in Electron you usually have to do an `npm rebuild`.  Try adding a rebuild npm command.  You'll need to change the target to your Electron version, and probably find the abi number (that's the nodejs protocol version).

```
"scripts": {
   ...
    "rebuild": "npm rebuild --runtime=electron --target=7.1.7 --disturl=https://atom.io/download/atom-shell --abi=75 && ./npmlink.sh"
  },
```