abandonware/noble #35: Display warning/error if running with insufficient permissions

oyooyo, 2019-12-12:
I just tried a minimalistic script for scanning the nearby peripherals, but somehow nothing happened at all. As it turned out, noble's state was and stayed "unknown" and would not switch to "poweredOn", even though my bluetooth dongle was attached.

With debug output turned on, I saw some messages suggesting there was a problem with insufficient permissions. After running the command shown in the "Running without root/sudo" section, everything worked fine.

If I remember correctly, the original noble displayed a helpful warning message if there was a problem with insufficient permissions.

rzr, 2019-12-12:
Hi, thanks for feedback,
can you report a version of original noble (and node version) and then eventually bisect to know when the regression was introduced ?

oyooyo, 2019-12-14:
I just tried to install the latest version of the original noble again. But apparently, the original noble version is now incompatible with my node version (v12.13.1) or something.

Anyway, I had a closer look at this issue and believe to have located the problem:
Lines 544+545 of file lib/hci-socket/hci.js are
```
if (error.message === 'Operation not permitted') {
  this.emit('stateChange', 'unauthorized');
```
but at least on my current system, error.message is not "Operation not permitted", but "EPERM, Operation not permitted". This causes hci.js to not recognize the error and not switch to the "unauthorized" state, which would cause noble to show the warning about insufficient permissions.

At first sight, I assume something like:
`if (error.message.includes('Operation not permitted')) {`
or
`if (error.code === 'EPERM') {`
might fix the issue.

https://github.com/abandonware/noble/blob/84967e50fdc77133f61a71b162e55e1e68cecfbd/lib/hci-socket/hci.js#L544

oyooyo, 2020-09-04:
Fixed via PR #111.

closed 2020-09-04 by oyooyo

rzr, 2020-09-04:
If needed you can release a version by mimicking recent "Release commits" :)

oyooyo, 2020-09-05:
@rzr I'm not 100% sure if I really understand what you mean by that (my experience with git is still a bit limited)

Do you mean I could create a PR with the changes that would be required for releasing a new version on npm? (which as I understand would come down to simply increasing the version number in package.json and package_lock.json)

rzr, 2020-09-05:
yes similarly to:

https://github.com/abandonware/noble/commit/c20a9b432da143cc96b8a9b152869e55f8980707

https://github.com/abandonware/noble/commit/9a26753fa32a7caf35acb5408cc5a1257a3913f9

Relate-to:

https://abandonware.github.io