abandonware/noble #36: install failing on Mojave (osx 10.14.6)

jakehawkes, 2020-01-14:
Hi!

First saw this compile error on mainline noble, and saw a comment there that suggested this fork. I am seeing the same compile failure:

    $ npm install @abandonware/noble
    
    > xpc-connection@0.1.4 install /Users/jake/code/node_modules/xpc-connection
    > node-gyp rebuild
    
    CXX(target) Release/obj.target/binding/src/XpcConnection.o
     ../src/XpcConnection.cpp:26:77: error: too few arguments to function call, single argument 'context' was not
      specified
    target->Set(Nan::New("XpcConnection").ToLocalChecked(), tmpl->GetFunction());
                                                          ~~~~~~~~~~~~~~~~~ ^
    /Users/jake/Library/Caches/node-gyp/13.6.0/include/node/v8.h:6283:3: note: 'GetFunction' declared here
    V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
    ^

Node v13.6.0
npm v6.13.4

Both freshly installed.

rzr, 2020-01-14:
Relate-to: https://github.com/abandonware/noble/issues/34

Any opensource mac developer here ?

jakehawkes, 2020-01-14:
I'm not an expert, but I've been in the OSS world for many years now. Perhaps there is something I could do?

rzr, 2020-01-14:
Well it's you to decide but maybe setting up CI for building on MacOS could help.

akx, 2020-02-02:
This seems to be a problem building [`xpc-connection`](https://github.com/sandeepmistry/node-xpc-connection), which probably hasn't been updated to new Node API standards.

I'm not immediately sure whether `xpc-connection` is a dependency of ours, but if it is, I think it'll also need the @abandonware fork treatment.

Meanwhile, @jakehawkes, can you attempt installing in an entirely new directory? `/Users/jake/code/node_modules` seems a little suspect :)

rzr, 2020-02-03:
Excellent  Thanks again for:

https://github.com/sandeepmistry/node-xpc-connection/pull/25#issuecomment-581408116

Feel free to try to suggest @sandeepmistry to relocate project in @abandonware/maintainers.

I think pingged him several times but I don't think I got real feedback.

Relate-to: https://github.com/abandonware/noble/issues/2