abandonware/noble #40: Using noble and bleno at the same time

RasPelikan, 2020-02-10:
Hi,

I want to build a proxy for a proprietary BLE gadget (health) providing a non-proprietary BLE service. At the moment a I can start advertising using bleno and scan for the gadget using noble. On connecting to the peripheral by noble the callback (https://github.com/abandonware/noble#connect) is never called and there is no error.

I use Windows 10 and WinUSB as a driver for an Intel Bluetooth 5.0 chip. I would like to use a Raspberry Pi Zero W for production (which I didn't buy yet).

1. There is a hint at https://github.com/abandonware/noble#bleno-compatibility about this, but this flag did not change anything. Might this be up to the WinUSB? Should WinUSB provide Bluetooth 4.1 functionality?
1. Should this work using the Raspberry Pi and Linux?
1. Should it work to connect to a peripheral as a central using noble and be connected to another centrals as a peripheral using bleno? Or is this something which won't work anyway?
1. Should this work using two processes? On windows it doesn't seem so, but what about Linux?
1. Is the only way to get this work by using two BLE adapters (e.g. Raspberry's onboard and another USB dongle)?

I appreciate for any hint

Cheers,
Stephan

rzr, 2020-02-11:
Hi,
I was able to use this cheap adapter on linux:

https://www.amazon.com/Bluetooth-Adapter-Computer-Desktop-Keyboard/dp/B07YH8H1PL/ref=rzr-21#

RasPelikan, 2020-02-13:
@rzr did you use Noble and bleno at the same time? Multirole is documented to be only possible for Bluetooth 4.1 and the dongle you posted is 4.0!

RasPelikan, 2020-02-20:
In the meantime I got the Raspberry Pi Zero W and I can report that it works! My program starts to advertise by using Bleno and then it scans for and connects to a peripheral by using Noble. I'm able to connect to the Bleno service during the active connection of Noble. So the Raspberry Pi acts as a kind of Bluetooth proxy.

I use the "NOBLE_MULTI_ROLE=1" setting but I could not see any difference.

In some circumstances I cannot connect to the Bleno service and I get an error starting with "noble: ..." although this should be handled by Bleno. But I have to investigate the situations which lead to this error and I will open another issue one I can reproduce this.

closed 2020-02-20 by RasPelikan