abandonware/noble #42: Noble crashes VS Code Extension host

pythoulon, 2020-02-17:
When I try to reference noble from a VS code extension, the VS code extension host crashes without any further message whenever a call to noble is made. Is this expected behavior ?
Small project to reproduce the problem is here : https://1drv.ms/u/s!AtFUKHonYn6jiapXDpfqcN7ohyagPA?e=PmNbYv

Thanks.

rzr, 2020-02-17:
May you try to setup an windows builder using github actions or any other bot that will build master branch ?

akx, 2020-02-17:
@pythoulon Which platform is this on? Do you have Bluetooth devices attached?

pythoulon, 2020-02-17:
@akx Sorry, this is on Windows 10 Fall Update (build 1909 if I'm not mistaken), VS Code latest version (1.42.1), noble v1.9.2-5.

pythoulon, 2020-02-17:
@rzr Not sure what you're referring to. I'm not very familiar with Github.

pythoulon, 2020-02-17:
@akx Forgot to mention : I'm using a TP-Link BT 4.0 USB dongle. It works fine as I'm able to run the [advertisement-discovery](https://github.com/abandonware/noble/blob/master/examples/advertisement-discovery.js) sample from this noble source tree.

pythoulon, 2020-02-17:
OK. I think I narrowed it down to this. The precompiled usb module that comes as part of the noble installation is apparently not compile against the proper Node version:

`Error: The module '\\?\c:\Users\pythm\OneDrive\Temp\noble-test-fork\node_modules\usb\build\Release\usb_bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
the module (for instance, using 'npm rebuild' or 'npm install').`

Neither `npm rebuild` nor `npm install` solve this. I'm going to investigate a bit more. Hints welcome.

pythoulon, 2020-02-18:
Finally got it. For some reason which you'll probably understand better than I do, the binaries have to be compiled against NODE_MODULE_VERSION 73 ABI, which is used by Electron v6, not node. So to get the thing to run, I had to:

1. Remove all node_modules (including clearing the npm cache)
2. install electron and electron-rebuild as dev dependencies
3. reinstall @abandonware/noble
3. force an electron-rebuild against version 6 of electron

The problem is with the usb module which is a dependency of bluetooth-hci-socket. Not sure whether this could be integrated to the noble install procedure... 

Apparently, this occurs in the VS code extension development environment, and not elsewhere (I was able to run the noble examples included with the package without having to do all this...).

akx, 2020-02-18:
Glad you were able to get to the bottom of it! :)

rzr, 2020-02-18:
May the issue be linked to usb module,
Are binaries published in module ?
If yes then maybe we should do this as well, and setup CI/CD flow for this,
this is what I tried to suggest earlier.

pythoulon, 2020-02-18:
@rzr The issue is indeed related to the usb module. As far as I understand, the binaries are built during the install process using gyp. Except they are built against the current node install (node 12 in my instance), but for some reason in my case, they had to be rebuilt against Electron V6. (is this a side effect of the VS Code extension build environment ? not sure, it's the first time I encounter this, but it's also the first time I work with a package that needs native recompilation...).

rzr, 2020-02-18:
Feel free to report to usb project and share link here.

pythoulon, 2020-02-18:
Report to node-usb here: https://github.com/tessel/node-usb/issues/350#issuecomment-587894703
(turns out I'm not alone ! 😉)

meghajayakumar, 2020-09-23:
@pythoulon I have the similar issue with node-usb whenever I use noble package. I tried the solution you have suggested above and rebuild it against electron version 6 but I still get the same issue. Any suggestions for the same? Thanks.

```
\electron\node_modules\usb\build\Release\usb_bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 73. This version of Node.js requires
NODE_MODULE_VERSION 80. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:769:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at bindings (C:\Users\Documents\AppDesktop\electron\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (C:\Users\Documents\AppDesktop\electron\node_modules\usb\usb.js:1:57)
```

pythoulon, 2020-09-23:
Yes. As of VSCode version v1.49, you must rebuild against Electron v9... 
The list of NODE_MODULE_VERSIONs vs Electron version number is maintained here :  https://github.com/nodejs/node/blob/master/doc/abi_version_registry.json

For the rebuild to work, you may also have to force installation of the latest electron (v10 something), electron-rebuild and  node-abi packages (had to do this when I switched to VSCode 1.49 last week...)

Good luck.