abandonware/noble #43: Using noble within a docker container

jackbridger, 2020-02-18:
I am trying to use noble in a node (v13.8.0) app running in a local docker container.

When I run my application I receive the below error. It seems like there is an issue with hci-socket.

`/app/node_modules/@abandonware/noble/lib/hci-socket/hci.js:75
  this._socket = new BluetoothHciSocket();
                 ^

Error: EAFNOSUPPORT, Address family not supported by protocol
    at new Hci (/app/node_modules/@abandonware/noble/lib/hci-socket/hci.js:75:18)
    at new NobleBindings (/app/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:28:15)
    at Object.<anonymous> (/app/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:511:18)
    at Module._compile (internal/modules/cjs/loader.js:1151:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1171:10)
    at Module.load (internal/modules/cjs/loader.js:1000:32)
    at Function.Module._load (internal/modules/cjs/loader.js:899:14)
    at Module.require (internal/modules/cjs/loader.js:1040:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at module.exports (/app/node_modules/@abandonware/noble/lib/resolve-bindings.js:13:12) {
  errno: 97,
  code: 'EAFNOSUPPORT',
  syscall: 'socket'
}
`

Does anyone have a workaround?

Thanks!

rzr, 2020-02-18:
I suspect problem in docker run params ?
Maybe we can add a docker-compose file

jackbridger, 2020-02-19:
These are the docker commands I'm using:

docker build -t mqtt-server .
docker run -it -p 9002:3002 -v $(pwd):/app mqtt-server /bin/bash

My Dockerfile:

`FROM node:latest
# 
WORKDIR specifies the directory our
# application's code will live within
WORKDIR /app
# We copy our package.json file to our
# app directory
RUN npm install -g nodemon
COPY package.json /app/package.json
# We then run npm install to install
# express for our application
RUN npm install && npm ls
RUN mv /app/node_modules /node_modules
# We then copy the rest of our application
# to the app direcoty
COPY . /app
# We start our application by calling
# npm start.

`
Package.json:

`
{
  "name": "TIG-docker",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "keywords": [],
  "author": "",
  "license": "ISC",
  "dependencies": {
    "@abandonware/noble": "^1.9.2-5",
    "mqtt": "^3.0.0"
  }
}

`

rzr, 2020-02-19:
Is the problem in mqtt or noble module ?

Maybe try to play along different network mode, I used:

https://github.com/rzr/webthing-iotjs/blob/master/docker-compose.yml#L21

jackbridger, 2020-02-19:
Thanks @rzr , Having done some more research, I think it's the bluetooth module in noble because I'm running docker locally on a mac but haven't found a solution so far?

rzr, 2020-02-19:
Sorry I am a linux user but I can give a try, may you open a PR with docker-compile file running a sample self test and I'll try it and will share you logs

C4mp3r-Grey, 2020-08-13:
Hello! As AcCalzone mentioned, I am running into the same phenomenon here https://github.com/AlCalzone/ioBroker.ble/issues/422#issue-677958385

I have closed it, as it does not seem to be an AlCalzone/BLE issue then.

Any update on this here?

C4mp3r-Grey, 2020-08-19:
Maybe this issue can be solved as I solved mine:

Run the container with _--net host_ rather than shared. 

Although this thread is pretty old, it may help others to find the own mistake as I did after several days? https://github.com/buanet/docker-iobroker/issues/134#issuecomment-676683897

rzr, 2020-09-06:
feel free to add a chapter in readme file about using with docker

atrovato, 2020-09-30:
Hi, did you run noble within a Docker container?
Can you share your Dockerfile, I'm not sure of required dependencies and commands to run.

SSharshunov, 2022-11-18:
> Я разработал свой собственный пакет npm, который зависит от`@abandonware/noble`
> 
> У меня похожая проблема. Вот ошибка:
> 
> `[node-red-contrib-ble-sense/protocols] Error: EAFNOSUPPORT, Address family not supported by protocol`
> 
> Я запускаю Node-RED на RPi 4, на котором работает DietPi.

Use ```--net=host ``` for docker cli, to resolve this problem (EAFNOSUPPORT)

ianchanning, 2024-04-14:
This is indeed now written up in the README > Docker section: https://github.com/abandonware/noble/#docker

Thanks to @atrovato :heart: