abandonware/noble #44: Is there a new release planned anytime soon ?

pythoulon, 2020-02-18:
Just asking because I need a version with type definitions included for typescript. The master branch head has that, but the latest release 1.9.2-5 does not. Thanks.

rzr, 2020-02-18:
This is possible, I am waiting reviews for:
https://github.com/abandonware/noble/pull/38

Can you please check those changes then I'll merge and release

Meanwhile you can use git url in npm manifests

pythoulon, 2020-02-18:
Well, I'm not really qualified to review https://github.com/abandonware/noble/pull/38, but I checked out the current head master and overwrote my current `node_modules/@abandonware/noble` (1.9.2.-5) with it, and it worked fine, for what it's worth... Can't really pretend this is an exhaustive test 😄

rzr, 2020-02-20:
Please double check this release and close issue:

Also please comment:
https://github.com/noble/noble/pull/851

pythoulon, 2020-02-24:
1.9.2-6 works fine for me.

closed 2020-02-24 by pythoulon