abandonware/noble #45: Fails to rebuild on electron 1.8.8!

herrjemand, 2020-02-23:
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
  CC(target) Release/obj.target/nothing/../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
In file included from ../lib/mac/src/noble_mac.mm:7:
In file included from ../lib/mac/src/noble_mac.h:3:
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:987:24: error: unknown type name 'napi_async_context'
                       napi_async_context context = nullptr) const;
                       ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:990:24: error: unknown type name 'napi_async_context'
                       napi_async_context context = nullptr) const;
                       ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:994:24: error: unknown type name 'napi_async_context'
                       napi_async_context context = nullptr) const;
                       ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:987:43: error: cannot initialize a parameter of type 'int' with an rvalue of type 'nullptr_t'
                       napi_async_context context = nullptr) const;
                                          ^         ~~~~~~~
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:987:43: note: passing argument to parameter 'context' here
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:990:43: error: cannot initialize a parameter of type 'int' with an rvalue of type 'nullptr_t'
                       napi_async_context context = nullptr) const;
                                          ^         ~~~~~~~
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:990:43: note: passing argument to parameter 'context' here
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:994:43: error: cannot initialize a parameter of type 'int' with an rvalue of type 'nullptr_t'
                       napi_async_context context = nullptr) const;
                                          ^         ~~~~~~~
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:994:43: note: passing argument to parameter 'context' here
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1016:7: error: unknown type name 'napi_deferred'
      napi_deferred _deferred;
      ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1168:30: error: unknown type name 'napi_async_context'
                             napi_async_context context = nullptr) const;
                             ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1171:30: error: unknown type name 'napi_async_context'
                             napi_async_context context = nullptr) const;
                             ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1175:30: error: unknown type name 'napi_async_context'
                             napi_async_context context = nullptr) const;
                             ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1168:49: error: cannot initialize a parameter of type 'int' with an rvalue of type 'nullptr_t'
                             napi_async_context context = nullptr) const;
                                                ^         ~~~~~~~
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1168:49: note: passing argument to parameter 'context' here
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1171:49: error: cannot initialize a parameter of type 'int' with an rvalue of type 'nullptr_t'
                             napi_async_context context = nullptr) const;
                                                ^         ~~~~~~~
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1171:49: note: passing argument to parameter 'context' here
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1175:49: error: cannot initialize a parameter of type 'int' with an rvalue of type 'nullptr_t'
                             napi_async_context context = nullptr) const;
                                                ^         ~~~~~~~
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1175:49: note: passing argument to parameter 'context' here
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1759:14: error: unknown type name 'napi_async_context'
    operator napi_async_context() const;
             ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:1763:5: error: unknown type name 'napi_async_context'
    napi_async_context _context;
    ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:2045:20: error: unknown type name 'napi_node_version'
      static const napi_node_version* GetNodeVersion(Env env);
                   ^
In file included from ../lib/mac/src/noble_mac.mm:7:
In file included from ../lib/mac/src/noble_mac.h:3:
In file included from /Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi.h:2051:
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi-inl.h:437:24: error: use of undeclared identifier 'napi_is_promise'
  napi_status status = napi_is_promise(_env, _value, &result);
                       ^
/Users/herrjemand/testElectronAPP/node_modules/node-addon-api/napi-inl.h:448:24: error: use of undeclared identifier 'napi_is_dataview'
  napi_status status = napi_is_dataview(_env, _value, &result);
[... 112 lines cut ...]
    at SafeSubscriber.error (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:156:26)
    at Subscriber._error (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/Users/herrjemand/testElectronAPP/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.next (/Users/herrjemand/testElectronAPP/node_modules/rxjs/internal/Subscriber.js:143:22)
Connecting process...
Error connecting to Noble API. The error is:  Error: The module '/Users/herrjemand/testElectronAPP/node_modules/noble/lib/mac/native/binding.node'
was compiled against the ABI-stable Node.js API (N-API).
This feature is experimental and must be enabled on the
command-line by adding --napi-modules.
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:172:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:172:20)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Module.require (module.js:513:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Users/herrjemand/testElectronAPP/node_modules/noble/lib/mac/bindings.js:4:16)
    at Object.<anonymous> (/Users/herrjemand/testElectronAPP/node_modules/noble/lib/mac/bindings.js:10:3)
ELECTRON VERSION: 1.8.8
NODE VERSION: 8.2.1

rzr, 2020-02-23:
Is 1.8.8 still supported ?

May you consider to use a supported stable release:

https://www.electronjs.org/releases/stable

It looks like there is no LTS:

https://www.electronjs.org/docs/tutorial/about

herrjemand, 2020-02-23:
@rzr Yeah, the project i am working on was stuck in 1.8.x due to library incompatibility, but after some work i have move all the way to 8.x XD

So thanks! XD

closed 2020-02-23 by herrjemand

rzr, 2020-02-23:
Is your project opensource ? If yes may a chapter be added on README file to track use of noble ?

herrjemand, 2020-02-23:
@rzr Sadly no.

But I can tell that any 1.8.8 has this issue

herrjemand, 2020-02-23:
@rzr it happens on the rebuild part