abandonware/noble #47: Is there any size limitations for notifications?

progerzua, 2020-02-26:
I`m subscribing for notification, that sends me 180 bytes of data, but size of data in callback is 20.
Is there any size limitations inside the library, that cuts data?

rzr, 2020-02-26:
https://github.com/abandonware/noble/pull/46 review welcome

progerzua, 2020-02-26:
@rzr thanks a lot! Looks like this is fix only for 'read', but not for 'notify' (I`ve cheked it on my system and still have the same issue).

I`ll try to do smth using your PR as example, but I`m not so good in js ¯\_(ツ)_/¯

camparijet, 2020-02-26:
@progerzua Honestly it's not clear enough to trace the cause. It would be much nicer if you can also share the dumped message by something like `btmon` via wireshark.

What's the output with `DEBUG="att hci"`? No related att messages? 

According to the spec (BLUETOOTH SPECIFICATION Version 5.0 | Vol 3, Part F 3.4.7.1 Handle Value Notification), 

> If the attribute value is longer than (ATT_MTU-3) octets, then only the first
(ATT_MTU-3) octets of this attributes value can be sent in a notification.
Note: For a client to get a long attribute, it would have to use the Read Blob
Request following the receipt of this notification.

This means, when the notification value is longer, the client also have to start sending Read Blob Request if it want to access the unsent part. Do you see kind of the request in the log?

And the socket data you put seems "Handle Value Indication", so maybe good idea to see the Handle Value Confirmation message from client.