abandonware/noble #49: Trouble Installing

ericdhkim, 2020-03-01:
Hi,

I'm trying to install noble, but running into an issue that I can't decipher. I'm rather ignorant about any kind of coding (trying to install noble to be able to install a homebridge plugin that requires it), so apologies if my question has an obvious answer.
I'm trying to install on a Raspberry Pi4 with Node.js version 13.9.0 and npm version 6.13.7.
Here's what I see when I run the install command

> sudo npm install @abandonware/noble

> bluetooth-hci-socket@0.5.1 install /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket
> node-gyp rebuild

gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/13.9.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp"
make: Entering directory '/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
In file included from ../src/BluetoothHciSocket.cpp:8:
../../../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
../src/BluetoothHciSocket.cpp: In static member function ‘static void BluetoothHciSocket::Init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
../src/BluetoothHciSocket.cpp:128:82: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
 get->Set(Nan::New("BluetoothHciSocket").ToLocalChecked(), tmpl->GetFunction());
                                                                             ^

In file included from /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/node.h:67,
                 from /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/v8.h:6283:46: note: candidate: ‘v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/v8.h:6283:46: note:   candidate expects 1 argument, 0 provided
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*’ is deprecated [-Wdeprecated-declarations]
 back(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                              ^

In file included from ../src/BluetoothHciSocket.cpp:8:
../../../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*’ is deprecated [-Wdeprecated-declarations]
 back(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                              ^

In file included from ../src/BluetoothHciSocket.cpp:8:
../../../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:269:109: error: no matching function for call to v8::Object::Get(v8::Local<v8::String>)’
 or = Local<Function>::Cast(globalObj->Get(Nan::New("Error").ToLocalChecked()));
                                                                             ^

In file included from /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/node.h:67,
                 from /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/v8.h:3594:43: note: candidate: ‘v8::MaybeLocal<v8::Value> v8::Object::Get(v8::Local<v8::Context>, v8::Local<v8::Value>)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                           ^~~
/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/v8.h:3594:43: note:   candidate expects 2 arguments, 1 provided
/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/v8.h:3597:43: note: candidate: ‘v8::MaybeLocal<v8::Value> v8::Object::Get(v8::Local<v8::Context>, uint32_t)’
   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
                                           ^~~
/home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/v8.h:3597:43: note:   candidate expects 2 arguments, 1 provided
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
In file included from /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/node.h:67,
                 from /home/pi/node_modules/noble-device/node_modules/bluetooth-hci-socket/.node-gyp/13.9.0/include/node/node_buffer.h:25,
[... 76 lines cut ...]
gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/13.9.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/@abandonware/noble/.node-gyp"
make: Entering directory '/home/pi/node_modules/@abandonware/noble/build'
  SOLINK_MODULE(target) Release/obj.target/noble.node
  COPY Release/noble.node
make: Leaving directory '/home/pi/node_modules/@abandonware/noble/build'
npm WARN pi@1.0.0 No description
npm WARN pi@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/noble-device/node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ @abandonware/noble@1.9.2-6
updated 1 package and audited 618 packages in 12.543s

2 packages are looking for funding
  run `npm fund` for details

found 2 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details

> 
Any help would be appreciated.

rzr, 2020-03-02:
Maybe we should use github actions to automate CI/CD.

rzr, 2020-03-05:
I automated some bits. it looks like node 13 is building fine
Please try current version to make sure

closed 2020-03-05 by ericdhkim

Zacknetic, 2020-08-16:
Hi, having the exact same issue on current version.