abandonware/noble #51: BLE IDs differ between OSX and Linux (Linux is correct)

mikeage, 2020-03-07:
Hi,
I'm a bit of a node.js newbie, so this bug report isn't directly on noble (but with a bit of guidance, I can get a more clear log). I'm using https://github.com/mKeRix/room-assistant on a raspberry pi and on my MacBook (running Mojave), and I get different IDs for my devices (Mi Band 4s and an Amazfit watch). On Linux, I get the proper IDs, however, on Mac, I get:
```
[Nest] 19083   - 03/07/2020, 12:17:48 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral 5015698a79934f2299c1089292f58244 with ID 5015698a79934f2299c1089292f58244 and RSSI -71
[Nest] 19083   - 03/07/2020, 12:17:52 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral 5deeee38f55d4b6ea55f2fa18950651c with ID 5deeee38f55d4b6ea55f2fa18950651c and RSSI -53
[Nest] 19083   - 03/07/2020, 12:17:53 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral dc2c15ea9978499d87cc9e18b27f6cd4 with ID dc2c15ea9978499d87cc9e18b27f6cd4 and RSSI -82
[Nest] 19083   - 03/07/2020, 12:18:02 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral 0e6c3f100ea940aaa3ee0cedcc4531bb with ID 0e6c3f100ea940aaa3ee0cedcc4531bb and RSSI -80
[Nest] 19083   - 03/07/2020, 12:18:22 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral a70d4e932254439093ba5298a73ede4b with ID a70d4e932254439093ba5298a73ede4b and RSSI -54
```
(These are the names if found and the values of `Peripheral.name` and `Peripheral.id`. I'm not sure what the convention is here, but it's clearly too long to be a MAC address.

Under Linux, these 5 devices are correctly identified as
```
Mar 01 16:03:01 raspberrypi room-assistant[25952]: [Nest] 25952   - 03/01/2020, 4:03:01 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral Mi Smart Band 4 with ID f66233931936 and RSSI -88
Mar 01 17:26:53 raspberrypi room-assistant[25952]: [Nest] 25952   - 03/01/2020, 5:26:53 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral Mi Smart Band 4 with ID e9b8fb1b6f63 and RSSI -87
Mar 01 17:48:11 raspberrypi room-assistant[25952]: [Nest] 25952   - 03/01/2020, 5:48:11 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral Mi Smart Band 4 with ID d11b4424249b and RSSI -85
Mar 01 19:59:47 raspberrypi room-assistant[25952]: [Nest] 25952   - 03/01/2020, 7:59:47 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral Mi Smart Band 4 with ID dd10a5420752 and RSSI -89
Mar 05 16:49:22 raspberrypi room-assistant[21185]: [Nest] 21185   - 03/05/2020, 4:49:22 PM   [BluetoothLowEnergyService] Discovered new BLE peripheral Amazfit Bip Watch with ID dce807d7d0e6 and RSSI -81
```
I'm not 100% sure which map to which.

Any ideas? I'd be happy to reproduce this with just noble if someone can give me a snippet that I an just to just do a scan and print the IDs and Names found.

mikeage, 2020-03-07:
(not sure if it matters, but this is node v13.10.1, installed from homebrew)

oeed, 2020-03-14:
I recall seeing that macOS does not show the real ID and instead generates its own UUID for the device - so likely not possible.