abandonware/noble #52: Issue with noble.on state change

ASHBAW, 2020-03-09:
I am trying to get this plugin working on a raspberry pi. I have an external BLE dongle attached to the pi. I have used the hcitool to verify that I can scan and connect to a BLE device.

I created a very simple node express application (node version - 10.19.0) and installed the plugin.

My package.json is as follows:

```
{
  "name": "angular-node-express-api",
  "version": "0.0.0",
  "private": true,
  "scripts": {
    "start": "node ./bin/www"
  },
  "dependencies": {
    "@abandonware/noble": "^1.9.2-7",
    "body-parser": "~1.13.2",
    "cookie-parser": "~1.3.5",
    "express": "~4.13.1"
  }
}
```

I copied the example code into my application as follows:

```
var noble = require('@abandonware/noble');

console.log('Noble ' + noble);

noble.on('stateChange', function(state) {
  console.log('Current Noble state ' + state);
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  console.log('peripheral discovered (' + peripheral.id +
              ' with address <' + peripheral.address +  ', ' + peripheral.addressType + '>,' +
              ' connectable ' + peripheral.connectable + ',' +
              ' RSSI ' + peripheral.rssi + ':');
  console.log('\thello my local name is:');
  console.log('\t\t' + peripheral.advertisement.localName);
  console.log('\tcan I interest you in any of the following advertised services:');
  console.log('\t\t' + JSON.stringify(peripheral.advertisement.serviceUuids));

  var serviceData = peripheral.advertisement.serviceData;
  if (serviceData && serviceData.length) {
    console.log('\there is my service data:');
    for (var i in serviceData) {
      console.log('\t\t' + JSON.stringify(serviceData[i].uuid) + ': ' + JSON.stringify(serviceData[i].data.toString('hex')));
    }
  }
  if (peripheral.advertisement.manufacturerData) {
    console.log('\there is my manufacturer data:');
    console.log('\t\t' + JSON.stringify(peripheral.advertisement.manufacturerData.toString('hex')));
  }
  if (peripheral.advertisement.txPowerLevel !== undefined) {
    console.log('\tmy TX power level is:');
    console.log('\t\t' + peripheral.advertisement.txPowerLevel);
  }

  console.log();
});
```
I can see an object is created from the first console.log, but the stateChange callback is never called (i.e. the state remains in an unknown state).

svogl, 2020-03-10:
did you run your program as root or use setcap on the node executable?

ASHBAW, 2020-03-10:
That was it. Running as root worked. Sorry where in the documentation is that?

And thank you!

rzr, 2020-03-10:
in the patch you're going to submit :)

ASHBAW, 2020-03-10:
Thanks svogl. I can't believe I missed that. My bad.

closed 2020-03-10 by ASHBAW