abandonware/noble #53: Handling USB reconnects

svogl, 2020-03-10:
We have an environment with lots of Wifi/BLE traffic that causes the Asus USB sticks to freeze every now and then. We work around this by reseting the stick (hciconfig hci0 down, up)

Unluckily, the hci.js does not recognize this but keeps the initial hci socket open. Besides complaining 'File descriptor in bad state', onSocketError does nothing useful.

As a test case (and preliminary fix), here is a minimal test case that scans for a second, forces the stick to power off, waits or it to appear again and re-initializes the hci layer with a new HciSocket object:
 
```
// survive restarts of usb dongles. initializes hciSocket again on error...
// tested with
// 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth

import { exec, execSync } from "child_process"

const noble = require('@abandonware/noble')

var BluetoothHciSocket = require('@abandonware/bluetooth-hci-socket');

noble._bindings._hci._socket.on('error', (error: any) => {
  console.log(error.message)
  // noble._bindings._hci._socket = new BluetoothHciSocket()
})

let timer: NodeJS.Timeout

function msleep (ms: number) {
  return new Promise(resolve => {
    setTimeout(resolve, ms)
  })
}

async function checkHciDevice () {
  let out = ''

  let up = execSync('hciconfig; hciconfig hci0 up; echo ""').toString()
  console.log('UP ' + up)
  if (up.indexOf('UP RUNNING')>0){
    msleep(200)
    console.log('hci found -- re-init hci!')
    noble._bindings._hci._socket = new BluetoothHciSocket()
    noble._bindings._hci.init()
    clearInterval(timer)
  } else {
    console.log('wait a little longer...')
  }
}

noble.on('stateChange', (state: string) => {
  console.log('noble--> stateChange ' + state)
  switch (state) {
    case 'poweredOn':
      noble.startScanning()
      // collect 1s of advertisments
      setTimeout(() => {
        execSync('hciconfig hci0 down; hciconfig')
      }, 1000)
      // after the timeout, the dongle disconnects and re-associates 
      break;
    case 'poweredOff':
      // wait for hci device to reconfigure and up it again. This needs a few seconds 
      // for the USB subsytem to reconfigure.
      setTimeout(()=>{
        timer = setInterval(checkHciDevice, 500)
      }, 5000)
      break;
    default:
      console.log('gaehn. ' + state)
  }
})

noble.on('discover', (peripheral: any) =>
  console.log(peripheral.address)
)
```

IMHO, onSocketError should at least emit an event for that case plus maybe a helper function to call hci.init() with a new socket object. Any suggestions?

akx, 2020-03-10:
A quick hackaround would be to just terminate your entire process on socket error and let a process manager restart it...

svogl, 2020-03-10:
@akx that's what we had but don't want :) we're tracking state of several sensors that we don't want to lose so auto-restart is the way to go for us

impala454, 2020-04-26:
I have been running the older noble library for years and have seen similar behavior.  I'll notice I've lost data from all my BLE advertisers I'm reading and only a down/up command solves it.  Restarting the entire node red process is not an option for reasons similar to what @svogl describes.   
   
@svogl this was over a month ago, have you been running your suggested fix?  I could definitely see this at least being an option in the node.

svogl, 2020-04-27:
@impala454 I've dug deeper into this - basically you need to reset every state-machine in every subsystem of the stack so that things start up again correctly - at least at noble, bindings, gap, hci and hcisocket modules are affected. I have a preliminary patch for this, but as we did not have too much time in our project I could not fully test it. 

Thus we have resorted to a more brutal approach by resetting the full noble subsystem. Basically you import noble as a modifiable variable
`let noble = require('@abandonware/noble')
`
and once you find out that it's stuck you reset it via
`
      noble = require('@abandonware/noble')
`
Works for us so far. But note that you lose any information collected in the meantime.

akx, 2020-04-27:
> Basically you import noble as a modifiable variable
> `let noble = require('@abandonware/noble') `
> and once you find out that it's stuck you reset it via
> `noble = require('@abandonware/noble')`
> Works for us so far. But note that you lose any information collected in the meantime.

Erm... that really shouldn't work at all due to the [require cache](http://nodejs.org/api/modules.html#modules_require_cache).

For proof, a trivial module being required twice only produces side effects once:

```
~/Desktop $ cat module.js
console.log("I am required.");                                                                                                                                                    ~/Desktop $ cat index.js
require('./module');
require('./module');                                                                                                                                                      ~/Desktop $ node index.js
I am required.
~/Desktop $
```

impala454, 2020-04-27:
I don't mind testing a potential fix as I have a system set up reading about 10 BLE advertisers 24-7.  I may dive into the code @svogl if you can share what you have so far.

svogl, 2020-04-27:
@akx honestly I was surprised that is actually does work. I was sceptical but gave it a try and it does re-open the hci socket to the newly enumerated USB device.
Still, this might be node version dependent; we're using 10.16.3

svogl, 2020-04-27:
@impala454 It's a bad hack really - I tried to fiddle with the noble state from the outside, as did not want to recurse over reset()-functions. So it's all in the recovery function ignoring any of the clean layer that have been introduced - just don't do it like this :)

```
// n.b. that we use typescript:
      // HCI bindings:
      noble._bindings._hci._socket = new BluetoothHciSocket()
      noble._bindings._hci._isDevUp = null
      noble._bindings._hci._state = null
      noble._bindings._state = null

      // reset GAP
      noble._bindings._gap._state = null
      noble._bindings._gap._scanState = null;
      noble._bindings._gap._scanFilterDuplicates = null;
      //noble._bindings._gap._discoveries = {};

      noble._bindings._addresses = {};
      noble._bindings._addresseTypes = {};
      noble._bindings._connectable = {};
      noble._bindings._pendingConnectionUuid = null;
      noble._bindings._connectionQueue = [];

      noble._bindings._handles = {};
      noble._bindings._gatts = {};
      noble._bindings._aclStreams = {};
      noble._bindings._signalings = {};

      noble._peripherals = {};
      noble._services = {};
      noble._characteristics = {};
      noble._descriptors = {};
      noble._discoveredPeripheralUUids = [];
      noble._state = null

      noble._bindings._hci.init()
```
just look at the constructors of say gatt.js etc to see if there are any more state variables that need resetting and maybe try to keep thte discovered objects at the noble layer.
S