abandonware/noble #74: index.d.ts error

finchlt, 2020-04-13:
node_modules/@abandonware/noble/index.d.ts:48:2 - error TS1036: Statements are not allowed in ambient contexts.
48 };

I can fix it by manually patching the file file to remove the semicolon at line 48. I don't think its an error per-se but my linter sure doesn't like it.

mrstegeman, 2020-04-13:
This was fixed in fef35bd53771ca72aef5e0a5e1d08ea471f9ec0a.

rzr, 2020-04-14:
But anyone can release a new version as suggested at :
https://github.com/abandonware/noble/commit/fef35bd53771ca72aef5e0a5e1d08ea471f9ec0a#commitcomment-38312011

knaos, 2020-05-13:
Well, why has nobody released that yet?

rzr, 2020-05-13:
not yet but you're free to do it, I'll merge

CyberCastle, 2020-05-13:
Ready. Pull request done.

rzr, 2020-05-14:
thanks please also review changes at:

https://github.com/noble/noble/pull/926