abandonware/noble #78: Does noble only work with USB bluetooth dongles? - No compatible USB Bluetooth 4.0 device found! (Windows Error)

dsteinman, 2020-04-23:
I'm using noble on a Macbook, my code works in MacOSX but gives the following error in Windows (boot camp install).

Based on this error, it looks like Noble actually requires a USB dongle, is that correct? 

Can anyone suggest a dirt-cheap dongle that is known to work in Windows?

```
C:\Users\Dan\bletest\node_modules\@abandonware\bluetooth-hci-socket\lib\usb.js:88
    throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\Dan\bletest\node_modules\@abandonware\bluetooth-hci-socket\
lib\usb.js:88:11)
    at BluetoothHciSocket.bindRaw (C:\Users\Dan\bletest\node_modules\@abandonware\bluetooth-hci-socket\l
ib\usb.js:44:8)
    at Hci.init (C:\Users\Dan\bletest\node_modules\@abandonware\noble\lib\hci-socket\hci.js:100:35)
    at NobleBindings.init (C:\Users\Dan\bletest\node_modules\@abandonware\noble\lib\hci-socket\bindings.
js:78:13)
    at C:\Users\Dan\bletest\node_modules\@abandonware\noble\lib\noble.js:60:24
    at processTicksAndRejections (intern
```

mrstegeman, 2020-04-23:
There are some listed in the README here: https://github.com/abandonware/node-bluetooth-hci-socket/blob/master/README.md

dsteinman, 2020-04-23:
Awesome, thank-you!

closed 2020-04-23 by dsteinman