abandonware/noble #79: Scanning silently stops working

AlCalzone, 2020-04-26:
I'm using this library in an app which scans for advertisements. I'm getting reports that this scan silently stops working after a while (timeframe of minutes to hours). As you can see in the log attached to https://github.com/AlCalzone/ioBroker.ble/issues/391#issuecomment-619495225, the scan generates a bunch of data every couple of milliseconds starting at `08:46:01.045` until `08:48:36.292`, then there is a sudden silence.

Do you guys have any clue how to start investigating this?

fascinated, 2020-07-01:
I am seeing this issue on OS X 10.15.5 / Mid-2014 MBP while scanning for advertisments either using this package directly, or via another wrapper with even the simplest code like [below]. It seems to happen after about an hour, but can also take longer.

`const BeaconScanner = require('node-beacon-scanner');
const scanner = new BeaconScanner();

scanner.onadvertisement = (ad) => {  
    console.log(JSON.stringify(ad, null, '  '));
};

// Start scanning
scanner.startScan().then(() => {
    console.log('Listening for beacons');
}).catch((error) => {
    console.error(error);
});`

Going to do some tests to see if any actions like exiting and restarting the node script after a certain interval or recreating the noble object at an interval helps; and then go from there. Hard to tell if it's an OS or library issue.

fascinated, 2020-07-01:
So far in my testing the issue seems to have gone away if I stop and restart scanning after, say ~10 min (it seems not necessary to restart the whole node script). Perhaps there is some kind of internal timeout at some point in the stack where the scanning stops if you've had it running too long or it has received too many messages? The advertising messages I am capturing arrive every ~20 sec, so this may all be different for you.

JsBergbau, 2020-11-28:
I have a similar issue with Python, so this problem is perhaps not related to noble. I use a watchdogtimer. If after X seconds no BLE packets are received, I stop scan and start again. That helps.
If I maintain several bluetooth connections and scanning is running it stops sometimes scanning after a few seconds. So I would say the more other processes access Bluetooth the more likely it is that scanning stops.

```python
def keepingLEScanRunning(): #LE-Scanning gets disabled sometimes, especially if you have a lot of BLE connections, this thread periodically enables BLE scanning again
	global BLERestartCounter
	while True:
		time.sleep(1)
		now = time.time()
		if now - lastBLEPaketReceived > args.watchdogtimer:
			print("Watchdog: Did not receive any BLE Paket within", int(now - lastBLEPaketReceived), "s. Restarting BLE scan. Count:", BLERestartCounter)
			disable_le_scan(sock)
			enable_le_scan(sock, filter_duplicates=False)
			BLERestartCounter += 1
			print("")
			time.sleep(5) #give some time to take effect
```
For fullcode see https://github.com/JsBergbau/MiTemperature2/blob/master/LYWSD03MMC.py