abandonware/noble #82: Node.js >= 12 on Raspberry Pi 3: state stuck on "unknown"

AlCalzone, 2020-05-02:
I've recently upgraded my Pi 3 to Node 14 and cannot get noble to scan anymore. The state is stuck on unknown and won't become "poweredOn" anymore. The same is reported for Node 12: https://github.com/AlCalzone/ioBroker.ble/issues/394

Interestingly, doing a manual scan on the console with `hcitool lescan` does change the state to poweredOn.

d-corler, 2020-05-08:
Same behavior with **pi user**, but works in **root**

oyooyo, 2020-05-22:
This is probably a duplicate of issue #35. If so, it comes down to the user not having sufficient permissions.

In such cases, noble is intended to print a warning message. But noble's current code for detecting such a "insufficient permissions" situation is unreliable; I assume the problem does not actually depend on the version of Node.js, but rather the version of the bluez library or so. Newer versions seem to emit a slightly different error message in that case, which makes noble fail to detect this situation.

Solution: Follow noble's instructions for solving the insufficient permissions problem on Linux (https://github.com/abandonware/noble#running-without-rootsudo-linux-specific)

AlCalzone, 2020-05-22:
@oyooyo this confirms my suspicions, thanks

closed 2020-05-22 by AlCalzone