abandonware/noble #91: (1967) uncaught exception: Cannot set property 'mtu' of undefined

r0b1zZle, 2020-05-28:
**Description**  
BLE Adapter is connecting well and consequently getting data from my flower sensors. Unfortunately every like 20 seconds the adapter is restarting due to an error.

**Expected behavior**  
No errors.

**Screenshots & Logfiles**  
`
host.raspberrypi4-iob | 2020-05-28 00:26:12.683 | info | Restart adapter system.adapter.ble.0 because enabled
-- | -- | -- | --
host.raspberrypi4-iob | 2020-05-28 00:26:12.683 | info | instance system.adapter.ble.0 terminated with code 0 (NO_ERROR)
host.raspberrypi4-iob | 2020-05-28 00:26:12.683 | error | Caught by controller[6]: noble: unknown peripheral 001a220754a6 handle notify!
host.raspberrypi4-iob | 2020-05-28 00:26:12.683 | error | Caught by controller[5]: at Hci.emit (events.js:315:20)
host.raspberrypi4-iob | 2020-05-28 00:26:12.683 | error | Caught by controller[5]: at NobleBindings.onAclDataPkt (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:288:15)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at AclStream.push (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/acl-stream.js:33:10)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at AclStream.emit (events.js:327:22)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at Gatt.onAclStreamData (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/gatt.js:133:26)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at Object.callback (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/gatt.js:329:10)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at Gatt.emit (events.js:315:20)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at NobleBindings.onMtu (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:277:8)
host.raspberrypi4-iob | 2020-05-28 00:26:12.682 | error | Caught by controller[5]: at NobleBindings.emit (events.js:315:20)
host.raspberrypi4-iob | 2020-05-28 00:26:12.681 | error | Caught by controller[5]: at Noble.onMtu (/opt/iobroker/node_modules/@abandonware/noble/lib/noble.js:564:18)
host.raspberrypi4-iob | 2020-05-28 00:26:12.681 | error | Caught by controller[5]: TypeError: Cannot set property 'mtu' of undefined
host.raspberrypi4-iob | 2020-05-28 00:26:12.681 | error | Caught by controller[4]: at Hci.emit (events.js:315:20)
host.raspberrypi4-iob | 2020-05-28 00:26:12.681 | error | Caught by controller[4]: at NobleBindings.onAclDataPkt (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:288:15)
host.raspberrypi4-iob | 2020-05-28 00:26:12.681 | error | Caught by controller[4]: at AclStream.push (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/acl-stream.js:33:10)
host.raspberrypi4-iob | 2020-05-28 00:26:12.681 | error | Caught by controller[4]: at AclStream.emit (events.js:327:22)
host.raspberrypi4-iob | 2020-05-28 00:26:12.680 | error | Caught by controller[4]: at Gatt.onAclStreamData (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/gatt.js:133:26)
host.raspberrypi4-iob | 2020-05-28 00:26:12.680 | error | Caught by controller[4]: at Object.callback (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/gatt.js:329:10)
host.raspberrypi4-iob | 2020-05-28 00:26:12.680 | error | Caught by controller[4]: at Gatt.emit (events.js:315:20)
host.raspberrypi4-iob | 2020-05-28 00:26:12.680 | error | Caught by controller[4]: at NobleBindings.onMtu (/opt/iobroker/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:277:8)
host.raspberrypi4-iob | 2020-05-28 00:26:12.680 | error | Caught by controller[4]: at NobleBindings.emit (events.js:315:20)
host.raspberrypi4-iob | 2020-05-28 00:26:12.680 | error | Caught by controller[4]: at Noble.onMtu (/opt/iobroker/node_modules/@abandonware/noble/lib/noble.js:564:18)
host.raspberrypi4-iob | 2020-05-28 00:26:12.679 | error | Caught by controller[4]: TypeError: Cannot set property 'mtu' of undefined
host.raspberrypi4-iob | 2020-05-28 00:26:12.679 | error | Caught by controller[3]: noble: unknown peripheral 001a220754a6 connected!
host.raspberrypi4-iob | 2020-05-28 00:26:12.679 | error | Caught by controller[2]: noble: unknown peripheral 001a220754a6 disconnected!
host.raspberrypi4-iob | 2020-05-28 00:26:12.678 | error | Caught by controller[1]: noble: unknown peripheral 001a220754a6 connected!
ble.0 | 2020-05-28 00:26:12.162 | info | (1967) Terminated (NO_ERROR): Without reason
ble.0 | 2020-05-28 00:26:12.161 | info | (1967) terminating
ble.0 | 2020-05-28 00:26:12.150 | error | (1967) TypeError: Cannot set property 'mtu' of undefined at Noble.onMtu (/opt/iobroker/node_modules/@abandonware/noble/lib/noble.js:564:18) at NobleBindings.emit (events.js:315:20) at Nobl
ble.0 | 2020-05-28 00:26:12.149 | error | (1967) uncaught exception: Cannot set property 'mtu' of undefined
ble.0 | 2020-05-28 00:26:12.143 | info | (1967) stopping scan
ble.0 | 2020-05-28 00:26:12.141 | error | (1967) TypeError: Cannot set property 'mtu' of undefined at Noble.onMtu (/opt/iobroker/node_modules/@abandonware/noble/lib/noble.js:564:18) at NobleBindings.emit (events.js:315:20) at Nobl
ble.0 | 2020-05-28 00:26:12.140 | error | (1967) uncaught exception: Cannot set property 'mtu' of undefined
`

**Versions:**  
 - Adapter version: 0.11.6
 - JS-Controller version: 2.2.10
 - Node / Nodejs version: v12.17.0
 - Operating system: Raspbian (Linux 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux)

rzr, 2020-05-28:
Can you please also compare with earlier versions ?

r0b1zZle, 2020-05-28:
> Can you please also compare with earlier versions ?

What do you mean? Which earlier versions?

AlCalzone, 2020-05-28:
@rzr @r0b1zZle The BLE adapter is using noble 1.9.2-8. Any specific versions we should compare with?

r0b1zZle, 2020-05-31:
Updated to js-controller: 3.1.4, still the same issue.

Fabryz, 2020-07-22:
My app just crashed with:
    "Cannot set property 'mtu' of undefined"
    /node_modules/@abandonware/noble/lib/noble.js:564
    peripheral.mtu = mtu;

Ubuntu Server 18.04, NodeJS 12.18.2, Noble 1.9.2-8.
This never happened in two months of developing with the library, so it's not easy to reproduce.
I just hope it won't crash during critical times

Patronics, 2020-10-26:
I also just got this error in noble (on node 14.10.1, but not on a similar system running node 12.19.0), did you ever find a solution?

Fabryz, 2020-10-26:
> I also just got this error in noble (on node 14.10.1, but not on a similar system running node 12.19.0), did you ever find a solution?

Nope, it didn't happen again, but also I had to work on another different project shortly after so I couldn't continue debugging....

fisherwise, 2021-01-25:
I had the same issue (#164) and I add a PR for it (#165) but in the meantime you can just apply my changes if you still have this issue

Jacksonbm1, 2021-05-09:
Sorry, can you clarify what changes you're referring to @fisherwise?