abandonware/noble #92: Example not working in node10/12

Junkern, 2020-05-28:
I noticed that while trying to develop with https://github.com/ChrisScheffler/miflora which uses this library. When I execute the quick start example (https://github.com/abandonware/noble#quick-start-example) with node v8 everything works wonderful. 

However, when I switch to node v10 or node v12 (always have to remove and reinstall this lib, so that everything gets complied correctly) it is stuck after the debug log `addressChange`.

It seems like the `stateChange` event is never triggered.

Log (I have two BLE devices):
```
DEBUG=* node noble2.js
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +1ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci read local version - writing: 01011000 +0ms
  hci write LE host supported - writing: 016d0c020100 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci read LE host supported - writing: 016c0c00 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci read bd addr - writing: 01091000 +0ms
  hci onSocketData: 040e0c01011000080000081d005a02 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4097 +0ms
  hci           status = 0 +0ms
  hci           result = 080000081d005a02 +0ms
  hci set scan enabled - writing: 010c20020001 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci set scan parameters - writing: 010b200701100010000000 +0ms
  hci onSocketError: EPERM, Operation not permitted +0ms
  hci onSocketData: 040e0a01091000fcdb99d0b69c +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4105 +1ms
  hci           status = 0 +0ms
  hci           result = fcdb99d0b69c +0ms
  hci address = 9c:b6:d0:99:db:fc +0ms
  noble addressChange 9c:b6:d0:99:db:fc +0ms
```

rzr, 2020-05-28:
it looks like a permission issue try again with sudo or adduser to the right group

Junkern, 2020-05-28:
Yeah, you were right. fixing permissions (https://github.com/abandonware/noble#running-without-rootsudo-linux-specific) with `sudo setcap cap_net_raw+eip $(eval readlink -f 'which node')` did the trick. My opinion: A short description on permissions right und er installation for linux would be helpful to avoid this for future users. I can make a PR, if you want.

closed 2020-05-28 by Junkern

rzr, 2020-05-28:
yes PR are welcome