abandonware/noble #93: Noble not working in Ubuntu 20.04

urstrulysufi, 2020-05-29:
I built a script reading advertisements from surrounding BLE devices on my Ubuntu 16.04 laptop. It worked perfectly fine. However, once I switched over to Ubuntu 20.04 I cannot discover any advertisements from surrounding BLE devices.
I tested with one of the examples in noble i.e. advertisement-discovery.js
I cannot discover any peripheral data.

Is there any way to workaround/fix this? Thanks.

dbochicchio, 2020-08-14:
Same here. Node app doesn't discover anything, while `hcisudo hcitool lescan` seems to correctly report devices.

oyooyo, 2020-08-30:
noble works on my system. I'm not using ubuntu but **k**ubuntu 20.04, but I don't think that makes a difference here. I'm using the latest LTS version of Node.js, v12.18.3, the latest @abandonware/noble version, and noble works fine. The mentioned example script advertisement-discovery.js works fine here as well.

Did you [set proper permissions for your user account](https://github.com/abandonware/noble#running-without-rootsudo-linux-specific) again after upgrading to 20.04? 
Because after switching to 20.04, I had to run the following command again:
```sh
sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)
```

Without doing so, noble did not work here as well. noble is intended to show a helpful hint about insufficient permissions in such situations, but because of #111, no error message gets shown when using newer versions of Node.js, nothing happens, and one might easily think that there is some problem with scanning or so.

chrwei, 2021-02-01:
I guess #111 was merged, 20.04 is working fine for me.  The instructions do say the setcap is required, error or not, but I think I did get the error.