abandonware/noble #95: Can not discover services and characteristics on devices.

p8nut, 2020-05-31:
i am currently working on a project which aim to control a Lego Duplo train with a [library](https://github.com/nathankellenicki/node-poweredup) that use noble.

after some research it seem that some device send services and characteristics before noble records them.

https://github.com/nathankellenicki/node-poweredup/issues/83

then i try to use the `peripheral-explorer.js`
here are the logs
```
node node_modules/@abandonware/noble/examples/peripheral-explorer.js c4:64:e3:2b:fa:4f 
peripheral with ID c464e32bfa4f found
  Local Name        = Train Base        
  Manufacturer Data = 9703002002fe4100
  Service Data      = []
  Service UUIDs     = 000016231212efde1623785feabcd123

services and characteristics:
```
and it wait forever.

here an extract of the logs with DEBUG=*
```

  hci onSocketData: 043e2b020100004ffa2be364c41f020106110723d1bcea5f782316deef12122316000009ff9703002002fe4100be +5s
  hci   event type = 4 +2ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +1ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 00004ffa2be364c41f020106110723d1bcea5f782316deef12122316000009ff9703002002fe4100be +0ms
  hci                   type = 0 +0ms
  hci                   address = c4:64:e3:2b:fa:4f +1ms
  hci                   address type = public +0ms
  hci                   eir = 020106110723d1bcea5f782316deef12122316000009ff9703002002fe4100 +0ms
  hci                   rssi = -66 +0ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[151,3,0,32,2,254,65,0]},"serviceData":[],"serviceUuids":["000016231212efde1623785feabcd123"],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +5s
  hci onSocketData: 043e2a020104004ffa2be364c41e1409547261696e2042617365002020202020202020051210002000020a00be +2ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 2 +0ms
  hci           LE meta event status = 1 +0ms
  hci           LE meta event data = 04004ffa2be364c41e1409547261696e2042617365002020202020202020051210002000020a00be +0ms
  hci                   type = 4 +1ms
  hci                   address = c4:64:e3:2b:fa:4f +0ms
  hci                   address type = public +0ms
  hci                   eir = 1409547261696e2042617365002020202020202020051210002000020a00 +0ms
  hci                   rssi = -66 +0ms
  gap advertisement = {"localName":"Train Base\u0000        ","txPowerLevel":0,"manufacturerData":{"type":"Buffer","data":[151,3,0,32,2,254,65,0]},"serviceData":[],"serviceUuids":["000016231212efde1623785feabcd123"],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]} +4ms
  hci set scan enabled - writing: 010c20020001 +1ms
peripheral with ID c464e32bfa4f found
  Local Name        = Train Base        
  Manufacturer Data = 9703002002fe4100
  Service Data      = []
  Service UUIDs     = 000016231212efde1623785feabcd123

services and characteristics:
  hci create le conn - writing: 010d20196000300000004ffa2be364c40006000c000000c80004000600 +10ms
  hci onSocketData: 040e04020c2000 +28ms
  hci   event type = 4 +1ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  noble scanStop +8s
  hci onSocketData: 040f040c010d20 +2ms
  hci   event type = 4 +0ms
  hci   sub event type = 15 +0ms
  hci           status = 12 +0ms
  hci           cmd = 8205 +1ms
```

hope you could help :-)
thanks

titaniumwhite, 2020-11-29:
@p8nut any solution so far?

p8nut, 2020-12-04:
haven't heard from anyone on this subject.
i may try again tonight and keep you updated.

titaniumwhite, 2020-12-04:
@p8nut I solved using the asynchronous version of the peripheral-explorer. 
It seems my device was sending the services and characteristics before the application was listening.

p8nut, 2020-12-13:
it's working now this is great thank's for your help

closed 2020-12-13 by p8nut