abandonware/noble #98: BLE Whitelist

rtborg, 2020-06-03:
Is there a way to configure a whitelist with Noble, similar to `hcitool lewladd`? I tried issuing the command `child_process.execSync('sudo hcitool lewladd xx:xx:xx:xx:xx:xx')` before `noble.startScanning([], true)`, but it did not have any effect.

My aim is to have the `discover` event triggered for devices in the white list only, and I should not have to check for MAC match manually.

JsBergbau, 2020-11-28:
I'd also like the ability to add an whitelist when you want to connect to a specific device.
Here is an example of hcidump when you connect via gatttool -I / connect FF:EE:DD:CC:BB:AA to a specific device. As you can see it first adds the specific device to whitelist and then starts scanning in the OS. FF:EE:DD:CC:BB:AA is the 

```
sudo hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0xffffffff
< HCI Command: LE Add Device To White List (0x08|0x0011) plen 7
> HCI Event: Command Complete (0x0e) plen 4
    LE Add Device To White List (0x08|0x0011) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x00 (passive)
    interval 60.000ms window 30.000ms
    own address: 0x00 (Public) policy: white list only
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr FF:EE:DD:CC:BB:AA (Public)
      Flags: 0x05
      Unknown type 0x07 with 16 bytes data
      RSSI: -61
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr FF:EE:DD:CC:BB:AA type 0
    interval 96 window 96 initiator_filter 0
    own_bdaddr_type 0 min_interval 24 max_interval 40
    latency 0 supervision_to 42 min_ce 0 max_ce 0
```

rabee05, 2021-07-03:
@rtborg Did you find out a way using noble to whitelist devices? Thanks!

rtborg, 2021-07-04:
> 
> 
> @rtborg Did you find out a way using noble to whitelist devices? Thanks!

Unfortunately, no.