abandonware/noble #103: Node.js Synchronous function calls (execution blocking)

philm400, 2020-06-11:
First off, I'm new to Node.js development and there may be a way to do this already, so might be more a case of Node.js/Noble dev for Noobs needed here

I'm trying to gather multiple static characteristics from a single service and store them in an object for reference directly after:
e.g.
```
let deviceInfo = {};
servs[0].discoverCharacteristics(null, function(error, chars) {
    chars[0].read((error, data) => {
        deviceInfo.model = data.toString();
        chars[1].read((error, data) => {
            deviceInfo.manufacturer = data.toString();
            chars[2].read((error, data) => {
                deviceInfo.hardware = data.toString();
                chars[3].read((error, data) => {
                    deviceInfo.firmware = data.toString();
                    chars[4].read((error, data) => {
                        deviceInfo.software = data.toString();
                        console.table(deviceInfo);
                    });
                });
            });
        });
    });
    if (deviceInfo.model == "ARC") { // do something }
});
```
Problem is, due to the non-blocking nature of Node.js function calls using callbacks, Node gets to my if() and executes it before the deviceInfo.model = data.toString(); line is executed in the first callback() function resulting in an undefined being returned.

I want to have the code execution blocked on each .read() until the data is returned and the object node variable set. E.g something like this:
```
deviceInfo.model = chars[0].read().toString();
deviceInfo.model = chars[1].read().toString();
deviceInfo.model = chars[2].read().toString();
deviceInfo.model = chars[3].read().toString();
deviceInfo.model = chars[4].read().toString();
if (deviceInfo.model == "ARC") { // do something }
```
Much easier to read, and waits for the function to complete until moving to the next.

There may be a way of doing this already using Promises and Promisify() but I can't figure it out myself it's all too new, and don't want to resort to altering core module dependancy code

I've also seen reference in the repo code to functions like readValue(). But I cannot find any documentation that details how to use these functions and what differentiates them from the likes of read()

philm400, 2020-06-11:
Answering my own question here but I think I've worked around my lack of Node.js knowledge and am using something like this to chain await Async function calls:

```
deviceInfo.model = (await (await serv[0].discoverCharacteristicsAsync(['2a24']))[0].readAsync()).toString();
        deviceInfo.manufacturer = (await (await serv[0].discoverCharacteristicsAsync(['2a29']))[1].readAsync()).toString();
        deviceInfo.hardware = (await (await serv[0].discoverCharacteristicsAsync(['2a27']))[2].readAsync()).toString();
        deviceInfo.software = (await (await serv[0].discoverCharacteristicsAsync(['2a28']))[3].readAsync()).toString();
        deviceInfo.firmware = (await (await serv[0].discoverCharacteristicsAsync(['2a26']))[4].readAsync()).toString();
```

it doesn't feel clean, but it's working.

Please do let me know if there is an easier cleaner way of doing this with Noble

chrwei, 2021-02-01:
the cleaner way is to use nodes Promise.all().  not using await means what's returned is a promise object, and when it's  resolved the .then() callback is called 

something like:
```
let calls = [];
calls.push(
	serv[0].discoverCharacteristicsAsync(['2a24']).then(c=> { //c is the characteristic array
		return c[0].readAsync(); //also no await, it returns a promise again, which chains another then()
	}).then(val => { //readAsync()'s result is now in val
		return { item: "model", value: val.toString() };
	})
);
calls.push(
	serv[0].discoverCharacteristicsAsync(['2a29']).then(c=> {
		return c[0].readAsync();
	}).then(val => {
		return { item: "manufacturer", value: val.toString() };
	})
);
//and so on
Promise.all(calls).then((results) => {
	//results is an array of the results from the last then() of each call.  
	results.forEach(r => {
		deviceInfo[r.item] = r.value;
	})

	if (deviceInfo.model == "ARC") { 
		// do something 
	}
});
```

the Promise.all().then() callback doens't ge called until each one of the async promises is resolved.

you can also streamline this some using implied returns in the lambdas, like:
```
calls.push(serv[0].discoverCharacteristicsAsync(['2a29']).then(c=> c[0].readAsync()).then(val => ({ item: "manufacturer", value: val.toString() })));
```