abandonware/noble #108: Hard node crash

kevinresol, 2020-07-22:
According to this comment: https://github.com/abandonware/noble/issues/12#issuecomment-484686555

Is there a reason why this happens and is it fixable? It is quite annoying because the js position information is lost and makes it very hard to debug.

tony-gutierrez, 2020-10-28:
Too vague. The issue you linked to is closed. Close this one.

kevinresol, 2020-10-28:
The issue is that error thrown in javascript causes a native crash (note the pointers to native functions in the original issue), and absent of javascript stack make it very hard to pin point the actual crash location in js

tony-gutierrez, 2020-10-28:
You aren't even mentioning what binding you are using, or any details on the error. This issue is too vague, and should be closed. There is no way the stack trace can be modified. You could always use a debugger.

kevinresol, 2020-10-28:
I don't think a javascript debugger can debug a hard crash at native level.

kevinresol, 2020-10-28:
and I will just copy the example code in case people failed to follow the link

```js
const noble = require("@abandonware/noble");
class Scanner {
  constructor() {
    noble.on("scanStart", this.onScanStart.bind(this));
    noble.on("stateChange", () => {
      noble.startScanning([], true);
    });
  }

  onScanStart() {
    throw 'error';
  }
}
const scanner = new Scanner();
```

```
FATAL ERROR: Error::Error napi_create_reference
 1: 0x100080c68 node::Abort() [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
 2: 0x100080dec node::errors::TryCatchScope::~TryCatchScope() [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
 3: 0x100080c7e node::OnFatalError(char const*, char const*) [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
 4: 0x10005e3c7 napi_open_callback_scope [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
 5: 0x105a01266 Napi::Error::Fatal(char const*, char const*) [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 6: 0x105a0731d Napi::Error::New(napi_env__*) [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 7: 0x105a14c06 Napi::FunctionReference::MakeCallback(napi_value__*, std::__1::vector<napi_value__*, std::__1::allocator<napi_value__*> > const&, napi_async_context__*) const [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 8: 0x105a146d0 ThreadSafeCallback::Impl::async_callback() [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 9: 0x105a1450e ThreadSafeCallback::Impl::static_async_callback(uv_async_s*) [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
10: 0x1008ba8a9 uv__async_io [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
11: 0x1008cb77f uv__io_poll [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
12: 0x1008bad36 uv_run [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
13: 0x1000b66fa node::NodeMainInstance::Run() [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
14: 0x10005dc1f node::Start(int, char**) [/Users/kevin/.nvm/versions/node/v12.16.1/bin/node]
15: 0x7fff676c8cc9 start [/usr/lib/system/libdyld.dylib]
16: 0x2 
[1]    6705 abort      node temp.js
```

```
$ node -v
v12.16.1
```

```
"@abandonware/noble": "^1.9.2-10"
```

tony-gutierrez, 2020-10-28:
You need to use a debugger to debug a debug build of the native code
binding, while running your node app, and importing the debug binding.

kevinresol, 2020-10-28:
I don't expect to debug at native level as a javascript library user. I am here to report the issue.

tony-gutierrez, 2020-10-28:
I recreated your code, and debugged using a simple javascript debuger, set to pause on any exception. It takes you right to the error line. 

[image]

kevinresol, 2021-05-27:
Same problem as at today:

macOS: 10.15.7
node: v14.5.0
@abandonware/noble: 1.9.2-13

```
~/Codes/test > node temp.js
FATAL ERROR: Error::Error napi_create_reference
 1: 0x1012762c5 node::Abort() (.cold.1) [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
 2: 0x1000a2899 node::Abort() [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
 3: 0x1000a29ff node::OnFatalError(char const*, char const*) [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
 4: 0x1000a28a9 node::FatalError(char const*, char const*) [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
 5: 0x10007965a napi_fatal_error [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
 6: 0x105401b7a Napi::Error::Fatal(char const*, char const*) [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 7: 0x10540813c Napi::Error::New(napi_env__*) [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 8: 0x105418ffd Napi::FunctionReference::MakeCallback(napi_value__*, std::__1::vector<napi_value__*, std::__1::allocator<napi_value__*> > const&, napi_async_context__*) const [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
 9: 0x1054189c7 ThreadSafeCallback::Impl::async_callback() [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
10: 0x1054187be ThreadSafeCallback::Impl::static_async_callback(uv_async_s*) [/Users/kevin/Codes/test/node_modules/@abandonware/noble/lib/mac/native/binding.node]
11: 0x10099f6fb uv__async_io [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
12: 0x1009b2a1a uv__io_poll [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
13: 0x10099fc81 uv_run [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
14: 0x1000e2687 node::NodeMainInstance::Run() [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
15: 0x100077742 node::Start(int, char**) [/Users/kevin/.nvm/versions/node/v14.5.0/bin/node]
16: 0x7fff73874cc9 start [/usr/lib/system/libdyld.dylib]
17: 0x2 
[1]    28401 abort      node temp.js
```