abandonware/noble #109: Memory leak

kevinresol, 2020-07-22:
The following script demonstrates the leak:

```js
setInterval(() => {
	var memory = process.memoryUsage();
	console.log(new Date().toString() + ': ' + (memory.heapUsed / 1024 / 1024 * 100 | 0) / 100 + " / " + (memory.heapTotal / 1024 / 1024 * 100 | 0) / 100 + " MB");
}, 5000);
	
var noble = require("@abandonware/noble")
noble.on("discover", peripheral => {
	console.log(peripheral.id);
});
noble.on("stateChange", status => {
	if(status == "poweredOn") {
		noble.startScanning([],true);
	}
});
```

On my macbook heap usage will steadily rise at a rate of around 0.2 MB/min, and will never go down.

I think this is might be the fix: https://github.com/noble/noble/pull/722/files

rzr, 2020-07-22:
Feel free to review upstream or rebase on @abandonware branch

kevinresol, 2020-07-22:
Sorry, I just ran this code for a longer time and noticed that GC will actually sweep once a while and memory usage is steady over a longer period. Closing for now.

closed 2020-07-22 by kevinresol

airdrummingfool, 2020-12-14:
I just ran a 1 month memory test using a slight modification of the code above, and I'm seeing a consistent 1.6mb/day increase in usage.
[image]

This was on an RPi 4B running on Node `14.15.0`. Over the course of the month, Noble called the `discover` event handler roughly 89 million times (there's about 20 constantly-advertising BLE devices within range).

@kevinresol how long did you run your test for? Would you consider re-opening this issue?

For the record, I don't think 600mb/**year** of memory bloat is in any way a crisis.