abandonware/noble #124: Unable to install package

meghajayakumar, 2020-09-23:
I am trying to install the package but its showing the below errors. Tried most of the solutions suggested online and still doesn't work.

Error message:

```
gyp ERR! UNCAUGHT EXCEPTION 
gyp ERR! stack Error: spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin ENOENT
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:19)
gyp ERR! stack     at onErrorNT (internal/child_process.js:415:16)
gyp ERR! stack     at process._tickCallback (internal/process/next_tick.js:63:19)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Ergotopia\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Ergotopia\Documents\Ergotopia\standingAppDesktop\node_modules\@abandonware\noble
gyp ERR! node -v v10.16.0
gyp ERR! node-gyp -v v7.1.0
gyp ERR! Node-gyp failed to build your package.
gyp ERR! Try to update npm and/or node-gyp and if it does not help file an issue with the package author.
```

Node version - v10.16.0
Gyp version  -  v7.1.0

rzr, 2020-09-23:
sorry no windows here,
but if you can help to add windows platform to CI/CD please do

Yongle-Fu, 2023-06-27:
same issue in macOS

Yongle-Fu, 2023-06-27:
process.nextTick change to setTimeout ?