abandonware/noble #126: Electron + Ionic - Node serialport and abandonware noble package in one app

meghajayakumar, 2020-09-25:
Summary of Problem
**- What I am trying to do?**
I am working on an desktop application with Ionic + Electron. I would like the application to have 2 options of connecting to a device, one through the serialport connection and other through a bluetooth connection using noble package.

**- What happens?**
The serialport functionalities are working good using serialport package. When I include the noble package for bluetooth functionalities I get a NODE_MODULE_VERSION error while building the project. After I fix the issue to make the app compatible with the noble package by re-building the project against the required electron version, the serialport package starts throwing the NODE_MODULE_VERSION.

**- What should have happened?**
I am expecting the project to be compatible with both the packages.

**Versions and Operating System**
Node.js v10.16.0
Windows

Flashs, 2020-09-29:
Hey,

i suggest you have a look at the electron docs: https://www.electronjs.org/docs/tutorial/using-native-node-modules . It has a nice description of what the underlying issue is and how to fix it.

Their suggestion is installing and running the electron-rebuild package. 
```npm install --save-dev electron-rebuild

# Every time you run "npm install", run this:
./node_modules/.bin/electron-rebuild

# On Windows if you have trouble, try:
.\node_modules\.bin\electron-rebuild.cmd```

megha24051994, 2020-09-30:
@Flashs  I have actually tried these. The noble package required an electron re-build against a particular version as well. I have tried the rebuild both the ways with these 2 packages and when the issue with one is fixed the other package starts throwing a NODE_MODULE_VERSION error.

meghajayakumar, 2020-09-30:
@Flashs  I have actually tried these. The noble package required an electron re-build against a particular version as well. I have tried the rebuild both the ways with these 2 packages and when the issue with one is fixed the other package starts throwing a NODE_MODULE_VERSION error.

Flashs, 2020-09-30:
@meghajayakumar 
I added serialport to my current project and running it gave me the same error. After running yarn electron-rebuild, i was able to use it just fine. Make sure to use the electron from the node_modules folder and not a system one.