abandonware/noble #128: too much services from discoverServices

lkurylo, 2020-10-05:
When I use the discoverServices with one uuid, I expect only one service in result. In many examples I see that, this is an expected behavior. So why I'm getting more that this one in which I'm interested in?

[image]

The same behavior I see in the characteristics.

I'm using @abandonware/noble with WinRT bindings, but with noble-winrt result is the same.
```
const { WinrtBindings } = require('noble-winrt/bindings.js');
var noble = require('noble/with-bindings')(new WinrtBindings());
```

tony-gutierrez, 2020-10-05:
See comment here: https://github.com/noble/noble/issues/950

pursual, 2020-10-28:
This is a binding problem, not a noble problem. A lot of the bindings do not respect the array that is passed in. They will discover all services, and sometimes filter that down, sometimes not. 

The proper behavior is to discover only the services in the array.