abandonware/noble #132: Error: ENODEV, No such device

atrovato, 2020-10-17:
Hello,
I work on a domotic application called Gladys: https://gladysassistant.com/

I build a integrated BLE service with noble package, but a user gets the following uncatchable error.

```
/src/server/services/bluetooth/node_modules/@abandonware/noble/lib/hci-socket/hci.js:100
    this._deviceId = this._socket.bindRaw(deviceId);
                                  ^

Error: ENODEV, No such device
    at Hci.init (/src/server/services/bluetooth/node_modules/@abandonware/noble/lib/hci-socket/hci.js:100:35)
    at NobleBindings.init (/src/server/services/bluetooth/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:78:13)
    at /src/server/services/bluetooth/node_modules/@abandonware/noble/lib/noble.js:60:24
    at processTicksAndRejections (internal/process/task_queues.js:79:11) {
  errno: 19,
  code: 'ENODEV',
  syscall: 'bind'
}
```

Does anybody know what it means?
Is there any way to not throw this king of error, even if BLE is not usable?

I'm a bit lost to explain this.

Thank you

michaelpalumbo, 2021-03-10:
Hi there, I am getting the same error, trying to run noble on an armv7 raspberry pi. 

```shell
/home/we/sto/node_modules/@abandonware/noble/lib/hci-socket/hci.js:120
    this._socket.bindRaw(this._deviceId);
                 ^

Error: ENODEV, No such device
    at Hci.init (/home/we/sto/node_modules/@abandonware/noble/lib/hci-socket/hci.js:120:18)
    at NobleBindings.init (/home/we/sto/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:93:13)
    at /home/we/sto/node_modules/@abandonware/noble/lib/noble.js:61:24
    at processTicksAndRejections (internal/process/task_queues.js:75:11) {
  errno: 19,
  code: 'ENODEV',
  syscall: 'bind'
}
```

stevenjoezhang, 2021-07-27:
I reproduced this problem on a Raspberry Pi 3b. It seems that the cause of the problem is that Bluetooth is not enabled properly. You can try to switch the Bluetooth on / off or reboot the device

fleutot, 2021-11-15:
I get this error if I try to access a HCI device which does not exist. For example, if `hcitool dev` gives me only `hci0`, but I try to access `hci1` this way, this is what I get:

```
$ DEBUG=* NOBLE_HCI_DEVICE_ID=1 nodejs discover-airglows.js
/home/gauthier/src/airglow_fw/tools/node/node_modules/@abandonware/noble/lib/hci-socket/hci.js:101
    this._deviceId = this._socket.bindRaw(deviceId);
                                  ^

Error: ENODEV, No such device
    at Hci.init (/home/gauthier/src/airglow_fw/tools/node/node_modules/@abandonware/noble/lib/hci-socket/hci.js:101:35)
    at NobleBindings.init (/home/gauthier/src/airglow_fw/tools/node/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:82:13)
    at Noble.<anonymous> (/home/gauthier/src/airglow_fw/tools/node/node_modules/@abandonware/noble/lib/noble.js:57:24)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3)
```
Maybe the systems where you see the error don't name their bluetooth unit `hci0`?

closed 2023-07-28 by rzr