abandonware/noble #136: startScanning doesn't do anything

meelash, 2020-11-02:
Hi,
I'm running noble on fedora, installed successfully, installed the prereqs and ran the setcap. Here is the code I'm running:
```
noble.on('stateChange', async (state) => {
  if (state === 'poweredOn') {
    console.log('starting scan')
    noble.startScanningAsync();
  }
})

noble.on('scanStart', () => {
  console.log('scanning started');
})

noble.on('discover', async (peripheral) => {
  console.log('found peripheral');
})
```
The only log output I see is `starting scan`, the scanStart event is not fired, neither are any peripherals discovered (I have tons of ble devices here that I can see using bluetoothctl or android apps, or the built-in gnome bluetooth settings window.

The output with DEBUG="*" is here:
```
  hci setting filter to: 1600000020c10000000000400000 +0ms
  hci set event mask - writing: 01010c08fffffbff07f8bf3d +0ms
  hci set le event mask - writing: 010120081f00000000000000 +1ms
  hci read local version - writing: 01011000 +0ms
  hci write LE host supported - writing: 016d0c020100 +0ms
  hci read LE host supported - writing: 016c0c00 +0ms
  hci read bd addr - writing: 01091000 +0ms
  hci onSocketData: 040e0401010c00 +4ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3073 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e0401012000 +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8193 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e0c010110000a00010a02000001 +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4097 +0ms
  hci           status = 0 +0ms
  hci           result = 0a00010a02000001 +0ms
  hci set scan enabled - writing: 010c20020001 +0ms
  hci set scan parameters - writing: 010b200701100010000000 +0ms
  hci onSocketData: 040e04016d0c00 +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3181 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e06016c0c000100 +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3180 +0ms
  hci           status = 0 +0ms
  hci           result = 0100 +0ms
  hci                   le = 1 +0ms
  hci                   simul = 0 +0ms
  hci onSocketData: 040e0a0109100043c7b7e8fc9c +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 4105 +0ms
  hci           status = 0 +0ms
  hci           result = 43c7b7e8fc9c +0ms
  hci address = 9c:fc:e8:b7:c7:43 +0ms
  noble addressChange 9c:fc:e8:b7:c7:43 +0ms
  hci onSocketData: 040e04010c200c +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010b200c +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8203 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
  noble stateChange poweredOn +10ms
starting scan
  hci set scan enabled - writing: 010c20020001 +1ms
  hci set scan parameters - writing: 010b200701100010000000 +0ms
  hci set scan enabled - writing: 010c20020101 +0ms
  hci onSocketData: 040e04010c200c +4ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010b200c +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8203 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
  hci onSocketData: 040e04010c200c +5ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 12 +0ms
  hci           result =  +0ms
```

 Any ideas?

Vishal8523, 2020-11-11:
Any updates on this issue ?
What is the work around to this ?

meelash, 2020-11-11:
My best guess, from the research I did was that it has something to do with hcictl being deprecated in recent linux in favor of bluetoothctl; noble is perhaps either using hcictl or using the same thing that hcictl is using and linux no longer allows that approach, which involves directly accessing the socket or something.

I'm not sure that's correct but that's the general conclusion I arrived at. I get very similar results trying to use hcictl to scan for ble devices, but with bluetoothctl I can scan, connect, read/write gatt parameters, the whole works.

Sorry, I don't have any workaround personally, I ended up going a completely different direction for what I needed to do.

chaeron, 2020-11-24:
I have exactly the same problem on the latest Ubuntu 20.04.1 OS.

Trying to use noble to find my new bangle.js watch (espruino device).  Neither the epruino --list command nor the code shown in the initial comment seem to find the bangle.js device, though I can see it using bluetoothctl or the bangle.js/espruino Web IDE.

chvolkmann, 2020-12-10:
Also confirmed for me at Ubuntu 20.04 1

chaeron, 2020-12-23:
I guess there is a reason this project is called "abandoware", eh? ;-)

rzr, 2020-12-23:
About abandonware:

https://rzr.github.io/rzr-presentations/docs/abandonware

is there any volunteers for doing better maintenance ?

chvolkmann, 2020-12-26:
@rzr I'd be willing to help out. That being said, I don't have too much experience with low level bluetooth stuff. If you could me some pointers in the right direction, I'll happily open a MR for this issue.

rzr, 2021-01-06:
I would compare trace between fedora, does it work as root ?

dragosprotung, 2021-02-08:
Strangely i managed to get it working inside a docker container (`FROM python:3.8`) on an Ubuntu machine, but not directly on the Ubuntu installation

chvolkmann, 2021-02-13:
@dragosprotung Hey! Could you explain your setup for using noble and attach a Dockerfile? Maybe we can work something out.

dragosprotung, 2021-02-19:
I am not sure if this will still work, as i do not have an Ubuntu machine to try it out

```
FROM ubuntu:18.04

RUN apt-get update && apt-get upgrade -y
RUN apt-get install -y bluetooth bluez libbluetooth-dev libudev-dev

RUN apt-get install -y curl xz-utils

RUN curl -sL https://deb.nodesource.com/setup_15.x | bash -
RUN apt-get install -y nodejs

# installs setcap
RUN apt-get install -y libcap2-bin
RUN setcap 'cap_net_raw+eip' $(eval readlink -f $(which  node))
RUN setcap 'cap_net_raw,cap_net_admin+eip' $(readlink -f $(which python3))
RUN setcap 'cap_net_raw+ep' $(readlink -f $(which hcitool))

RUN mkdir /app
COPY ./package.json /app

WORKDIR /app

RUN apt-get install -y build-essential
RUN npm install @abandonware/noble
RUN npm install node-switchbot

COPY ./test.js /app
```

TATOAO, 2023-03-10:
Ubuntu 22.04.2 LTS, I have exactly the same issue, noble startSanning(), but no peripherals were found, does anyone make it?