abandonware/noble #138: Should have clear documentation that Python 2.7 is required

panzer, 2020-11-06:
Improvement request
On Mac OSX 10.14.6, node v13.6.0, npm 6.13.4

Ran into this error trying to install

```
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
Traceback (most recent call last):
  File "/Users/..../node_modules/@abandonware/noble/build/./gyp-mac-tool", line 611, in <module>
    sys.exit(main(sys.argv[1:]))
  File "/Users/..../node_modules/@abandonware/noble/build/./gyp-mac-tool", line 30, in main
    exit_code = executor.Dispatch(args)
  File "/Users/..../node_modules/@abandonware/noble/build/./gyp-mac-tool", line 45, in Dispatch
    return getattr(self, method)(*args[1:])
  File "/Users/..../node_modules/@abandonware/noble/build/./gyp-mac-tool", line 248, in ExecFilterLibtool
    if not libtool_re.match(line) and not libtool_re5.match(line):
TypeError: cannot use a string pattern on a bytes-like object
make: *** [Release/nothing.a] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Users/..../.nvm/versions/node/v13.6.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:321:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Darwin 18.7.0
gyp ERR! command "/Users/..../.nvm/versions/node/v13.6.0/bin/node" "/Users/..../.nvm/versions/node/v13.6.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/..../node_modules/@abandonware/noble
gyp ERR! node -v v13.6.0
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@^0.5.3-5 (node_modules/@abandonware/noble/node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-6: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @abandonware/noble@1.9.2-10 install: `node-gyp rebuild`
npm ERR! Exit status 1
```

I eventually figured out this was caused by my local `python` install being Python 3. Downgrading to Python 2.7 fixed this issue. Should be better documented so others do not waste time dealing with this

Thanks to the contributors for upkeep on this project!

hmmhmmhm, 2022-07-03:
I had this problem after updating my Mac to Montrey. Python2 has been deleted from the latest Mac and requires manual installation

donavanbecker, 2022-07-03:
This didn’t work for me