abandonware/noble #139: Unable to connect ble devices on MAC 10.15.7 (Catalina)

yusheng-jia, 2020-11-09:
It can discover the ble sensors, but when call 
`peripheral.connect( function(error)
{
...
}
`
No output on the callback function.

rubengarciaalba, 2021-02-28:
Hi @yusheng-jia, did you find the solution for that?

I'm trying to run the quick start example on MacOS Catalina 10.15 but nothing happens after calling "startScanningAsync" function.

Node version is: 8.9.0
Npm version is: 5.5.1

yusheng-jia, 2021-03-02:
> Hi @yusheng-jia, did you find the solution for that?
> 
> I'm trying to run the quick start example on MacOS Catalina 10.15 but nothing happens after calling "startScanningAsync" function.
> 
> Node version is: 8.9.0
> Npm version is: 5.5.1

Hi I have fixed it on firmware side.()

yusheng-jia, 2021-03-02:
It maybe the issue of macOS . Please see https://github.com/xsens/xsens_dot_server/issues/16 .

closed 2021-03-02 by yusheng-jia