abandonware/noble #142: Peripheral stays in state disconnecting after disconnect

martbrugg, 2020-11-13:
I currently have a problem that the bluetooth device does not disconnect properly when it is already powered off.

1. I start a connection to a bluetooth device subcscribe to services an characteristics
2. The bluetooth device has lost power.
3. Call disconnect on the periperal but the callback is never called.
4. The peripheral stays in disconnecting state.

Is there a possibilite to force a disconnect on a peripheral.

[image]

angelmerino, 2021-02-09:
I'm having the same issue when stopping the connection before it gets connected.
I use a timeout if connection to the device takes too long but also the user can cancel the connection. I have played with cancelConnection (#120) but the peripheral gets stuck in disconnecting status and no way to get it out.

disconnect and connect never returns and the bluetooth interface will stop working. Then a hci reset is needed to get things working again.

Any idea where to look in order to fix this issue?

angelmerino, 2021-03-02:
HI, an small update.
I have seen that when I call disconnect (disconnectAsync in my case) the ongoing connection actually gets cancel but when the disconnect event should be sent, the peripheral is not available. I'm getting "noble warning: unknown handle 0 disconnected!"

I have tracked it to here in lib/noble.js
`
Noble.prototype.onDisconnect = function (peripheralUuid) {
  const peripheral = this._peripherals[peripheralUuid];

  if (peripheral) {
    peripheral.state = 'disconnected';
    peripheral.emit('disconnect');
  } else {
    this.emit('warning', `unknown peripheral ${peripheralUuid} disconnected!`);
  }
};
`

This is why the peripheral keeps in disconnecting state and the disconnect event never comes.
When and why the handle is gone, is still not clear.

rzr, 2021-03-02:
Relate-to: https://github.com/abandonware/noble/pull/173

jaulz, 2021-03-30:
@rzr sorry, why do you think it is related? This one is about disconnecting and the PR is about connecting. Maybe I just don't see the missing piece...