abandonware/noble #143: Noble does not work after upgrading to MacOS Big Sur

TSonono, 2020-11-13:
This piece of code results in node giving me "5262 abort      node dist/ble_script.js". This only happens after I upgraded to the latest MacOS version (11.0.1).

```typescript
        noble.on('stateChange', (state: string) => {
            if (state === 'poweredOn') {
                noble.startScanning([], false);
            } else {
                process.exit(0);
            }
```

TSonono, 2020-11-13:
Nevermind, this was due to me running the script while upgrading xcode.

virtualxdriver, 2020-11-18:
Could this have anything to do with Bluetooth entitlements?

https://stackoverflow.com/questions/64829740/prompting-for-bluetooth-permission-on-bug-sur-11-0-1

TSonono, 2020-11-19:
Yes you're right @virtualxdriver! I was able to solve this by going in to to System Preferences -> Security & Privacy -> Privacy and adding iTerm (or I guess Terminal if you're not using iTerm) in the Bluetooth section. Thanks!

closed 2020-11-19 by TSonono