abandonware/noble #152: Unable to use Noble => Error: EAFNOSUPPORT, Address family not supported by protocol

MrMaxime, 2020-11-30:
Hello, I'm on ubuntu terminal on windows, I tried to install @abandonware/noble, the installation succeed but when I try to launch my program, the program stop at my first line : `const noble = require('@abandonware/noble');`
I have this error : 

```
/mnt/c/Users/maxim/OneDrive/Documents/Cours ESIR/Cours ESIR2/Projet IoT/groupe-1bis-homedeck-christophe-maxime-paul/testNoble/node_modules/@abandonware/noble/lib/hci-socket/hci.js:74
  this._socket = new BluetoothHciSocket();
                 ^

Error: EAFNOSUPPORT, Address family not supported by protocol
    at new Hci (/mnt/c/Users/maxim/OneDrive/Documents/Cours ESIR/Cours ESIR2/Projet IoT/groupe-1bis-homedeck-christophe-maxime-paul/testNoble/node_modules/@abandonware/noble/lib/hci-socket/hci.js:74:18)
    at new NobleBindings (/mnt/c/Users/maxim/OneDrive/Documents/Cours ESIR/Cours ESIR2/Projet IoT/groupe-1bis-homedeck-christophe-maxime-paul/testNoble/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:25:15)
    at Object.<anonymous> (/mnt/c/Users/maxim/OneDrive/Documents/Cours ESIR/Cours ESIR2/Projet IoT/groupe-1bis-homedeck-christophe-maxime-paul/testNoble/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:546:18)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at module.exports (/mnt/c/Users/maxim/OneDrive/Documents/Cours ESIR/Cours ESIR2/Projet IoT/groupe-1bis-homedeck-christophe-maxime-paul/testNoble/node_modules/@abandonware/noble/lib/resolve-bindings.js:13:12) {
  errno: 97,
  code: 'EAFNOSUPPORT',
  syscall: 'socket'
}
```
```
node -v 
v12.20.0
npm -v
6.14.8
```
I tried to uninstall and reinstall after, but nothing change. I was previously in the latest version of Node.js and downgrade it but nothing change too.
I don't know how to solve it, if you have an idea, I will gladly hear it.
Thanks in advance.

jpalo, 2021-01-22:
Seeing this too on RasPi 4B.

```
node -v
v10.23.1
npm -v
6.14.10
```

mattjbones, 2021-03-02:
I had the same issue with my PiZero W (using DietPi) - turns out I didn't have the bluetooth adapter enabled (posting on the off chance this sparks some inspiraton). 

To diagnose I tried running the bluetooth normally, without going via Noble using `bluetoothctl` - is should return the MAC address of the bluetooth adapter and then allow for device discovery etc. However, when I ran the command it simply timed out. 

Whilst digging in the `dietpi-config` and I noticed that Bluetooth was turned off. So I enabled it there, rebooted my Pi and then ran `systemctl start hciuart`. Running `bluetoothctl` now scans bluetooth devices.

mattjbones, 2021-03-03:
I had the same issue with my PiZero W (this is using DietPi). 

I tried running the bluetooth normally, without going via Noble by `bluetoothctl` - is should return the MAC address of the bluetooth adapter. However, when I ran the command it simply timed out. 

Whilst digging in the `dietpi-config` and I noticed that Bluetooth was turned off. So I enabled it there, rebooted my Pi and then ran `systemctl start hciuart`. Running `bluetoothctl` now scans bluetooth devices and the Noble basic test (scan devices and get the battery status) works a treat!

jpalo, 2021-03-05:
Thanks @mattjbones, I can confirm this is due to BT not running correctly, will investigate and get back to how I managed to fix it.

```
hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-03-05 08:54:42 EET; 8s ago
  Process: 4013 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Mar 05 08:54:40 openhab systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Mar 05 08:54:42 openhab btuart[4013]: Failed to read local name, invalid HCI event
Mar 05 08:54:42 openhab btuart[4013]: Can't initialize device: Success
Mar 05 08:54:42 openhab btuart[4013]: bcm43xx_init
Mar 05 08:54:42 openhab systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Mar 05 08:54:42 openhab systemd[1]: hciuart.service: Failed with result 'exit-code'.
Mar 05 08:54:42 openhab systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.

```

jpalo, 2021-03-05:
So my issue was related to mismatch Raspi firmware version and modules, so I am booting from SD to USB (SSD), and the `uname -a` showed for the SD boot version

```
[10:11:38] openhabian@openhab:~$ uname -a
Linux openhab 5.4.83-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```
but corresponding modules were missing from SSD:
```
[10:15:06] openhabian@openhab:~$ modprobe bluetooth
modprobe: FATAL: Module bluetooth not found in directory /lib/modules/5.4.83-v7l+
```
but they were in `/lib/modules/5.10.17-v7l+`

So, modified /etc/fstab to boot from SD, excecuted `sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel` to also update SD firmware files, and restarted.

Summary: As this isn't Noble issue, I think this issue can be closed.