abandonware/noble #162: `Abort trap: 6` error when running on MacOSX Big Sur (11.1)

taktran, 2021-01-16:
I've been able to install on MacOSX Big Sur (11.1), but when I run it, I get the following error: `Abort trap: 6`

Any ideas why?

I've got

* `XCode Version 12.3`
* Tried it on nodejs `v10.22.1`, `v12.20.1` and `v14.9.0`, but same error

I thought it might be related to https://github.com/bazelbuild/rules_nodejs/issues/1958#issuecomment-678942630, but the nodejs versions I've installed seem to have the patch

Might also be related to https://github.com/abandonware/noble/issues/159, but not sure why I need to run `sudo npm install -g --unsafe-perm @abandonware/noble`

---

Installation

```
$ node --version
v12.20.1
$ npm i

> @abandonware/noble@1.9.2-10 install /Users/ttt/Work/Dev/rubik/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/src/nothing.o
  LIBTOOL-STATIC Release/nothing.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: archive library: Release/nothing.a the table of contents is empty (no object file members in the library define global symbols)
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
  CXX(target) Release/obj.target/binding/lib/mac/src/napi_objc.o
  CXX(target) Release/obj.target/binding/lib/mac/src/ble_manager.o
  CXX(target) Release/obj.target/binding/lib/mac/src/objc_cpp.o
  CXX(target) Release/obj.target/binding/lib/mac/src/callbacks.o
  SOLINK_MODULE(target) ../lib/mac/native/binding.node
  COPY Release/binding.node
  SOLINK_MODULE(target) Release/noble.node
npm WARN rubik@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@0.5.3-6 (node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-6: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})

added 5 packages from 56 contributors and audited 100 packages in 28.507s

6 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

Code I'm running (from [Quick Start example](https://github.com/abandonware/noble#quick-start-example))

```
// From Quick Start example (https://github.com/abandonware/noble#quick-start-example)
const noble = require("@abandonware/noble");

console.log("start");

noble.on("stateChange", async (state) => {
  console.log("stateChange", state);
  if (state === "poweredOn") {
    await noble.startScanningAsync(["180f"], false);
  }
});

noble.on("discover", async (peripheral) => {
  console.log("discover");
  await noble.stopScanningAsync();
  await peripheral.connectAsync();
  const {
    characteristics,
  } = await peripheral.discoverSomeServicesAndCharacteristicsAsync(
    ["180f"],
    ["2a19"]
  );
  const batteryLevel = (await characteristics[0].readAsync())[0];

  console.log(
    `${peripheral.address} (${peripheral.advertisement.localName}): ${batteryLevel}%`
  );

  await peripheral.disconnectAsync();
  process.exit(0);
});
```

Output

```
$ node index.js
start
Abort trap: 6
```

egiljae, 2021-01-19:
Try adding your terminal app to "System Preferences" —> "Security & Privacy" —> "Bluetooth".

taktran, 2021-01-19:
That was it, thanks @egiljae !

closed 2021-01-19 by taktran

rzr, 2021-01-19:
may you patch documentation  ?

HappyStriker, 2021-01-20:
> Try adding your terminal app to "System Preferences" —> "Security & Privacy" —> "Bluetooth".

Thanks, this works like a charm when working on the Terminal directly. But which entry do I need to add if I access my Mac via SSH?

Thanks in advance.

dnschnur, 2021-01-23:
The instructions in the updated docs only work when running from a local terminal. As @HappyStriker pointed out, when SSHed into the machine one still gets "Abort trap: 6'.

What binary needs to be added to the list when logged in via SSH? I tried adding node, sshd, bash, launchd, and rebooting my machine in case the permission doesn't apply retroactively. None of those worked.

garragames, 2023-01-13:
> Try adding your terminal app to "System Preferences" —> "Security & Privacy" —> "Bluetooth".

Thanks a lot, this save me!!