abandonware/noble #169: Bluetooth Updates on Raspberry PI 4 in Homebridge.

Foxtrot79, 2021-02-04:
I am not sure if this is the right place or not but I am using a plugin in my Home Bridge on my raspberryPI 4 that uses this plugin and it was suggested by that creator to reach out here for assistancee.  The plugin I am using is homebridge-plugin-govee.

The issue that I am seeing is that when the PI is powered on and Homebridge starts updates arrive from my devices and are updated in the accessories section of Homebridge.  After about 10 minutes these updates seem to cease even though the devices are still advertising (able to see with Govee app in iOS).

I am not sure if the issue is here or with something deeper.  I thought I would start here.

Thank you for any help.

ne0nex, 2021-04-17:
I'm having the same or similar issue with the miflora plugin. After a while the updates seem to stop. Performing an hcitool lescan (the devices are BTLE) at the time of the failure results in an I/O error to the bluetooth interface. Bouncing the state of the device using hciconfig hci0 reset fixes it for a few hours. Something about the way BT is working on my raspberry pi 4 with noble does this for some reason. Could it be possible to add BT interface state checking to noble so that if the interface is in an I/O error state, it gets reset? For now I wrote a shell script that resets the device every four hours but that solution seems hardly ideal.

Zephalon, 2022-10-03:
I have the same issue with my Govee thermometer in Node Red on my Pi 4. After some time there are no more updates.

Restarting the BT adapter and service did not fix it for me.