abandonware/noble #170: Installing Noble on Alpine Linux - Dependencies

carlosanunes, 2021-02-06:
The instructions do not specify how to install the library in Alpine, that is the required dependencies or equivalent ones for the libraries defined for other linux distributions.

rzr, 2021-02-06:
isnt alpine supporting npm or yarn ?

carlosanunes, 2021-02-07:
My questions is on the dependencies. Alpine Linux uses apk ( https://wiki.alpinelinux.org/wiki/Alpine_Linux_package_management ).

carlosanunes, 2021-02-07:
For Alpine I found bluez, libudev-**zero** and libudev-**zero**-dev ( to replace libudev ), but no bluetooth library ( is this an actual requirement ). Is this enough for an actual working state ?

carlosanunes, 2021-02-07:
Managed to install without errors, just warning but now experiencing same issue as #152

carlosanunes, 2021-02-20:
Found that installing bluez, libudev-zero and libudev-zero-dev packages are enough to met Noble dependencies in Alpine.

Cheers.

closed 2021-02-20 by carlosanunes