abandonware/noble #186: TypeError: require(...) is not a constructor on macOS Catalina?

Patronics, 2021-03-03:
Running macOS Catalina 10.15.7 and noble with both node 12 and node 14 fails with this error: 

```
patrickleiser$ node bluetooth.js 
/Users/username/path/to/project/node_modules/@abandonware/noble/lib/resolve-bindings.js:11
    return new (require('./mac/bindings'))(options);
           ^

TypeError: require(...) is not a constructor
    at module.exports (/Users/username/path/to/project/node_modules/@abandonware/noble/lib/resolve-bindings.js:11:12)
    at Object.<anonymous> (/Users/username/path/to/project/node_modules/@abandonware/noble/index.js:2:51)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/Users/username/path/to/project/bluetooth.js:7:15)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
```

peterlcsteele, 2021-03-04:
Same issue experienced on Mojave.

ralpioxxcs, 2021-03-04:
I have same problem too.
* 10.15.6 macOS Catalina
* noble 1.9.2-12

rzr, 2021-03-04:
regression might come from @atrovato's change:

https://github.com/abandonware/noble/commit/743b97551708ffc1392dc65d3f80db39238b8ac6#diff-5f8789720a0cb32cd50bd1f2ac87089febed143de711df08eb6cd610d44bb9c1R11

I wish it would have been triggered by CI , please can you try improve tests ?

atrovato, 2021-03-04:
I will, asap.

atrovato, 2021-03-04:
@rzr testing will be difficult, please see https://github.com/abandonware/noble/pull/188/checks?check_run_id=2034662987

atrovato, 2021-03-04:
Ok, PR is ready, I do tests with proxyquire to simulate all envs.