abandonware/noble #187: OS X: require is not a constructor

lutzbickhardt, 2021-03-04:
> @abandonware/noble@1.9.2-12 install /Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/nothing.o
  LIBTOOL-STATIC Release/nothing.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: archive library: Release/nothing.a the table of contents is empty (no object file members in the library define global symbols)
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
  CXX(target) Release/obj.target/binding/lib/mac/src/napi_objc.o
  CXX(target) Release/obj.target/binding/lib/mac/src/ble_manager.o
  CXX(target) Release/obj.target/binding/lib/mac/src/objc_cpp.o
  CXX(target) Release/obj.target/binding/lib/mac/src/callbacks.o
  SOLINK_MODULE(target) ../lib/mac/native/binding.node
  COPY Release/binding.node
  SOLINK_MODULE(target) Release/noble.node
npm WARN saveError ENOENT: no such file or directory, open '/Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/package.json'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @abandonware/bluetooth-hci-socket@^0.5.3-7 (node_modules/@abandonware/noble/node_modules/@abandonware/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for @abandonware/bluetooth-hci-socket@0.5.3-7: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN enoent ENOENT: no such file or directory, open '/Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/package.json'
npm WARN marked-terminal@4.1.1 requires a peer of marked@^1.0.0 || ^2.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN neu_node No description
npm WARN neu_node No repository field.
npm WARN neu_node No README data
npm WARN neu_node No license field.

+ @abandonware/noble@1.9.2-12

...
"
HERE: resolve-bindings" from console.log>

TypeError: require(...) is not a function
    at module.exports (/Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/node_modules/@abandonware/noble/lib/resolve-bindings.js:12:40)
    at Object.<anonymous> (/Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/node_modules/@abandonware/noble/index.js:2:51)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/explore_all_ble_devices_async.js:109:15)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
(base) lutzbickhardt_27@Lutzs-iMac neu_node % node --version
v14.15.1
(base) lutzbickhardt_27@Lutzs-iMac neu_node % sw_vers
ProductName:	Mac OS X
ProductVersion:	10.15.7
BuildVersion:	19H524
(base) lutzbickhardt_27@Lutzs-iMac neu_node % 
```
same program works without this problem under Ubuntu 20.04, node.js 10.19.0

I am lost here, too many "require"s chained to understand what is going on ...

lutzbickhardt, 2021-03-04:
the require was:
const noble = require('@abandonware/noble');

atrovato, 2021-03-04:
Dupe with #186

closed 2021-03-09 by rzr

rzr, 2021-03-10:
you can release npm package by opening a PR like mine, please also check builder testers logs

lutzbickhardt, 2021-03-10:
?i have to correct myself I mean #190 is still open, not #188

Cordialement

--------------------------
Lutz BICKHARDT
lutz@bickhardt.fr

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

________________________________
De : Bickhardt, Lutz
Envoyé : mercredi 10 mars 2021 14:03
À : abandonware/noble
Objet : RE: [abandonware/noble] OS X: require is not a constructor (#187)

sorry to correct, I believe you mean:

 Closed #18<https://github.com/abandonware/noble/issues/187>6 and #187 via #18<https://github.com/abandonware/noble/pull/188>7.

#188 is still open.

You have to download manually (e.g. via git clone). NPM is not updated yet.

Cordialement

--------------------------
Lutz BICKHARDT
lutz@bickhardt.fr

Graumannsweg 42, D-22087 Hamburg, Germany

SKYPE: bickhardt

+49 162 9784296
+49 172 3846086
+49 40 2275812

Closed #187<https://github.com/abandonware/noble/issues/187> via #188<https://github.com/abandonware/noble/pull/188>.

-

PR #188 (2021-03-09): Fix mac dependancy ctor
Fix #187 and #188