abandonware/noble #190: noble cannot be installed under OS X 10.13. 6 (High Sierra)

lutzbickhardt, 2021-03-06:
when checking for workarounds of issue 186/187 I tried an older iMac and using nvm checked node.js versions 8, 10, 12, 14, 15;

all ended by producing
```
lutzbickhardt$ npm install  @mapbox/node-pre-gyp
removed 34 packages, and audited 244 packages in 2s
found 0 vulnerabilities

iMac-215:~ lutzbickhardt$ npm install @abandonware/noble
npm WARN deprecated node-pre-gyp@0.17.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm ERR! code 1
npm ERR! path /Users/lutzbickhardt/node_modules/@abandonware/noble
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! CC(target) Release/obj.target/nothing/../../node-addon-api/nothing.o
npm ERR!   LIBTOOL-STATIC Release/nothing.a
npm ERR! Usage: /usr/local/bin/libtool [OPTION]... [MODE-ARG]...
npm ERR! Try 'libtool --help' for more information.
npm ERR! gyp info it worked if it ends with ok
...

```

the issue seems to be LIBTOOL-STATIC (see next post), because on a newe OS X it seems to ok.

Fixing the problem would be nice since older OS Xes cannot upgrade to higher than High Sierra, so one would be stuck here.

lutzbickhardt, 2021-03-06:
here it works on Catalina:

lutzbickhardt_27@Lutzs-iMac neu_node % npm install @abandonware/noble            

> @abandonware/noble@1.9.2-12 install /Users/lutzbickhardt_27/Documents/PlatformIO/Projects/bluetooth_server/neu_node/node_modules/@abandonware/noble
> node-gyp rebuild

  CC(target) Release/obj.target/nothing/../../node-addon-api/nothing.o
  LIBTOOL-STATIC Release/nothing.a
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/libtool: archive library: Release/nothing.a the table of contents is empty (no object file members in the library define global symbols)
  CXX(target) Release/obj.target/binding/lib/mac/src/noble_mac.o
  CXX(target) Release/obj.target/binding/lib/mac/src/napi_objc.o
  CXX(target) Release/obj.target/binding/lib/mac/src/ble_manager.o
  CXX(target) Release/obj.target/binding/lib/mac/src/objc_cpp.o
  CXX(target) Release/obj.target/binding/lib/mac/src/callbacks.o

stormbytes, 2021-03-29:
I'm on Catalina. Cannot get it to install either. 
See https://github.com/abandonware/noble/issues/199#issue-843845261