abandonware/noble #194: Write characteristic

rubenfsantiago, 2021-03-14:
I'm trying to connect a micro:bit with a NodeJS server using BLE connection and i want receive accelerometer data.  i get to connect, and i get data from accelerometer without problem.

Here is the problem and i hope get solution here. I want accelerometer data with a period of 640ms.  Then, microbit has a characteristic _e95dfb24251d470aa062fa1922dfa9a8_ with valid values: 1, 2, 5, 10, 20, 80, 160 and 640.  But when i write buffer data i do not observe change and receive a few of data in a second when i need only 2 datas in a second.

Here is my code, and i hope you can help me. 

P.d. sorry for my english ;-)

```
const ACCELEROMETERSERVICE_SERVICE_UUID = 'e95d0753251d470aa062fa1922dfa9a8';
const ACCELEROMETERDATA_CHARACTERISTIC_UUID = 'e95dca4b251d470aa062fa1922dfa9a8';
const ACCELEROMETERPERIOD_CHARACTERISTIC_UUID = 'e95dfb24251d470aa062fa1922dfa9a8';

noble.on('discover', peripheral => {
  console.log(
    `Found device, name: ${peripheral.advertisement.localName}, uuid: ${peripheral.uuid}`
  );

  if (peripheral.uuid === 'XXXXXXXXXXX') {
    noble.stopScanning();
    peripheral.on('connect', () => console.log('Device connected'));
    peripheral.on('disconnect', () => console.log('Device disconnected'));

    peripheral.connect(error => {
        var serviceUUIDs = [ACCELEROMETERSERVICE_SERVICE_UUID, BUTTON_SERVICE_UUID];
        
        peripheral.discoverServices(serviceUUIDs, (error, services) =>{
        console.log(`Found service name: ${services[0].uuid}`); 
        console.log(`Found service name: ${services[1].uuid}`); 
        const service =services[0];
        const serviceButton =services[1];

        const acelerometroUUIDs = [ACCELEROMETERDATA_CHARACTERISTIC_UUID, ACCELEROMETERPERIOD_CHARACTERISTIC_UUID]

        service.discoverCharacteristics(acelerometroUUIDs, (error, characteristics)=>{
                const datos=characteristics[0];
                const periodo = characteristics[1];
               var  milisegundos = Buffer(Uint16Array.from([640]).buffer)
            
                periodo.write (milisegundos,false)
                
                datos.on('data', (data)=>{
                        datos.read((err,readData) =>{
                                var buffer = readData;
                                var xUpdate =  buffer.readInt16LE(0);
                                var yUpdate = buffer.readInt16LE(2);
                                console.log(" x: "+xUpdate+ "  y: "+yUpdate);
                        });
                });
                datos.subscribe(() =>{
                        console.log("SUSCRITO A LA CARACTERISTICA ACELEROMETRO ");
                });
        });//service.discoverCharacteristics

(.....)
```

ujjwal-neu, 2022-08-10:
Hey @rubenfsantiago did you find any solution for this?

rubenfsantiago, 2022-08-10:
No, i didn't

RootDev4, 2024-08-20:
For me, set writing without response to false did the trick (macOS Sonoma 14.6.1)
`await characteristic.writeAsync(Buffer.from('abc', 'ascii'), false)`