abandonware/noble #202: Input Device

ThisLightMan, 2021-04-26:
Hello, I have two different Bluetooth input devices. Noble is not recognizing these devices unfortunately. I've tried https://www.npmjs.com/package/node-bluetooth, and it can successfully find both bluetooth input devices, but unfortunately, it isn't maintained. Is there something I'm doing wrong for these devices to not show on Noble?

Thanks!

airdrummingfool, 2021-04-28:
Are these devices "normal" Bluetooth, or do they support Bluetooth Low Energy (BLE)? If they are traditional Bluetooth input devices (mouse, keyboard, etc) they likely don't support BLE. noble only interfaces with BLE devices.

ThisLightMan, 2021-04-28:
Thanks for the response!

I'm not sure how to tell if the devices support BLE, but that would make sense. Is there a way for me to check?

If I confirm they do not support BLE, do you have any suggestions on how I could connect to the devices and get their output? I know noble won't work for this, but I wasn't sure if you had any other ideas in mind, no worries if you don't.

Thanks for the help

airdrummingfool, 2021-04-28:
If they don't show up in noble, I'm pretty sure they don't support BLE 😉 

More seriously, you could check the product's packaging, manual, and system requirements to see if they mention BLE or Bluetooth Low Energy. The "Bluetooth version" can even be used as a hint - BLE is only supported in Bluetooth 4.0 and newer, so if your device is "Bluetooth 2.0" or "Bluetooth 3.0" then it is definitely not BLE.

I don't do any work with non-BLE devices, so unfortunately I can't help you there. Good luck!

ThisLightMan, 2021-04-28:
Ah, yep, you're right. The device is Bluetooth 3.0.

Thanks again for the help!

closed 2021-04-28 by ThisLightMan